learnsyslab/gym-pybullet-drones #2: Installation of the conda environment for windows

shanki98, 2020-09-14:
**Is your feature request related to a problem? Please describe.**
The file/conda_req_list is setup for OSx

**Describe the solution you'd like**
is it possible to upload a files/conda_req_list for win x64 ? I

JacopoPan, 2020-09-14:
hi @shanki98 

I do not have a windows machine at hand right now but the only additional packages (not included by `conda create`) should be
```
pip install gym
pip install pybullet
pip install 'ray[rllib]'
```
and `pip install stable-baselines3` for `examples/learn.py`

let me know if you have any issue (possibly include the terminal output)

shanki98, 2020-10-07:
Hey @JacopoPan , I was able to run the simulator on windows. I tried running the file "fly.py", which uses PID Control, But I just see the quadrotor floating in air . Why is this and what should I do?. Parallelly, I also tried installing it in Ubuntu 16.04 but somehow I keep getting the modulenotfound error for "gym_pybullet_drones". You had said you tested it for ubuntu. Can you maybe upload a required package list for ubuntu? that would be very helpful.

JacopoPan, 2020-10-07:
Hi @shanki98 

Please go to the repository folder and run `git pull` to make sure you are using the most up to date code

Script `fly.py` should contain 3 drones flying on a circular trajectory at different altitudes 

With regard to the “module not found” error, make sure you have installed the package with `pip install -e .`

shanki98, 2020-10-07:
This is my error now in the Ubuntu environment. I cloned the repository today. Could you tell me what seems to be the problem @JacopoPan 
pybullet build time: Sep 22 2020 00:56:01
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Segmentation fault (core dumped)

JacopoPan, 2020-10-07:
This my output on Uvuntu 18, it seems that you have a SEGFAULT just before `Created GL 3.3 context`
```
(base) jacopo@Computer:~/gym-pybullet-drones/examples$ python fly.py 
pybullet build time: Jul  8 2020 18:24:12
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=Quadro P2000/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 450.66
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 450.66
Vendor = NVIDIA Corporation
Renderer = Quadro P2000/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
```
I'd say the problem is with PyBullet failing to create an OpenGL context.
Look at this issue https://github.com/bulletphysics/bullet3/issues/1681
Are you using a virtual machine?
If the problem is with OpenGL3, you could try to change line 103 in `BaseAviary.py` to
```
self.CLIENT = p.connect(p.GUI, options="--opengl2")
```

shanki98, 2020-10-07:
@JacopoPan  Yes, I'm using vmware (virtual machine) inside my laptop to run Ubuntu 16.04. 
I still get this output
pybullet build time: Sep 22 2020 00:56:01
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
argv[0]=--opengl2
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=3
argv[0] = --unused
argv[1] = --opengl2
argv[2] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
4

	visual 0x21 selected
Segmentation fault (core dumped)

shanki98, 2020-10-07:
When I tried it on windows I was able to get the rendering but as I mentioned the drone does not move. Ignore the last error because I closed the connection.
pybullet build time: Sep 14 2020 14:51:33
WARNING:tensorflow:From C:\Users\shash\anaconda3\lib\site-packages\tensorflow\python\compat\v2_compat.py:96: disable_resource_variables (from tensorflow.python.ops.variable_scope) is deprecated and will be removed in a future version.
Instructions for updating:
non-resource variables are not supported in the long term
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
starting thread 0
started testThreads thread 0 with threadHandle 0000000000000250
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 451.67
Vendor = NVIDIA Corporation
Renderer = GeForce MX250/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 0000000000000E0C
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
C:\Users\shash\anaconda3\lib\site-packages\gym\logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))

[INFO] BaseAviary.render() ——— it 0001 ——— wall-clock time 1.6s, simulation time 0.0s@240Hz (0.00x)
[INFO] BaseAviary.render() ——— drone 0 ——— x +00.00, y +00.00, z +00.10 ——— velocity +00.00, +00.00, -00.04 ——— roll +00.00, pitch -00.00, yaw +00.00 ——— angular velocities +00.00, +00.00, +00.00 ———

[INFO] BaseAviary.render() ——— it 0241 ——— wall-clock time 13.6s, simulation time 1.0s@240Hz (0.07x)
[INFO] BaseAviary.render() ——— drone 0 ——— x -00.10, y -00.03, z +00.10 ——— velocity -00.16, -00.07, +00.00 ——— roll +00.42, pitch +00.74, yaw -00.03 ——— angular velocities +12.73, -15.06, -02.03 ———

[INFO] BaseAviary.render() ——— it 0481 ——— wall-clock time 25.6s, simulation time 2.0s@240Hz (0.08x)
[INFO] BaseAviary.render() ——— drone 0 ——— x -00.25, y -00.13, z +00.10 ——— velocity -00.12, -00.15, -00.00 ——— roll +01.05, pitch -00.01, yaw -00.01 ——— angular velocities +16.38, -03.90, -01.61 ———
numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 0000000000000E0C exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 0000000000000250 exiting
Thread TERMINATED
Traceback (most recent call last):
  File "examples/fly.py", line 58, in <module>
    obs, reward, done, info = env.step(action)
  File "c:\tue\internship\crazyflie simulator\gym-pybullet-drones-master\gym_pybullet_drones\envs\BaseAviary.py", line 188, in step
    self._showDroneFrame(i)
  File "c:\tue\internship\crazyflie simulator\gym-pybullet-drones-master\gym_pybullet_drones\envs\BaseAviary.py", line 487, in _showDroneFrame
    self.Y_AX[nth_drone] = p.addUserDebugLine(lineFromXYZ=[0,0,0], lineToXYZ=[0,AXIS_LENGTH,0], lineColorRGB=[0,1,0], parentObjectUniqueId=self.DRONE_IDS[nth_drone], parentLinkIndex=-1, replaceItemUniqueId=int(self.Y_AX[nth_drone]), physicsClientId=self.CLIENT)
pybullet.error: Not connected to physics server. % ignore this one
[image]

JacopoPan, 2020-10-07:
are you simply running 
```
python fly.py
```
or are you passing any argument, e.g.
```
python fly.py --num_drones 1
```

shanki98, 2020-10-08:
@JacopoPan Yes, it now seems to work with proper rendering in OpenGL 3 in Windows. But the rotor doesn't seem to be moving as you show in the GIF. It is tracking the circular trajectory. Further, I also wanted to ask you this question that I am trying to implement MPC on this, is it enough if I just replace the PID controller with MPC, or should I change any other modules?

JacopoPan, 2020-10-08:
Yes, the gif is from an older commit, the tracked (planar) trajectory is the `TARGET_POS = np.zeros((NUM_WP,3))` array whose values are initialized on line 54
https://github.com/JacopoPan/gym-pybullet-drones/blob/d2b08fa49334e93db512c5ef5b1993ab18bcc555/examples/fly.py#L54

For the second point, I imagine that you will want to modify or replace the call to `computeControlFromState`, which produces rotor speeds given the current `state=obs[str(j)]["state"]` and `target_pos`
https://github.com/JacopoPan/gym-pybullet-drones/blob/d2b08fa49334e93db512c5ef5b1993ab18bcc555/examples/fly.py#L81

closed 2020-10-09 by JacopoPan

shanki98, 2020-10-12:
Hey @JacopoPan , I didn't parse any arguments through the interactive shell, I simply changed the num_drones to 1 instead of 3