learnsyslab/gym-pybullet-drones #3: ValueError in _updateAndStoreKinematicInfo

hooverryan, 2020-09-18:
I have a fresh install of the repo.  

I tried running the examples/fly.py file.  The GUI opened, everything initialized, then got an error in _updateAndStoreKinematicInfo during the first step.

I am running on Windows.

```
self.pos[i], self.quat[i] = p.getBasePositionAndOrientation(self.DRONE_IDS[i], physicsClientId=self.CLIENT)
ValueError: setting an array element with a sequence.
numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 0000000000000D44 exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 00000000000002A4 exiting
Thread TERMINATED
```

hooverryan, 2020-09-18:
I inadvertently changed the shape of self.pos in the _housekeeping function.  As a result, the iteration through the drones in _updateAndStoreKinematicInfo was creating an error.  This was my error.

closed 2020-09-18 by hooverryan

JacopoPan, 2020-09-18:
Thanks @hooverryan 
```
python fly.py
```
the script works fine on macOS and Ubuntu, can you check if replacing _updateAndStoreKinematicInfo with this
```
    def _updateAndStoreKinematicInfo(self):
        for i in range (self.NUM_DRONES):
            #self.pos[i], self.quat[i] = p.getBasePositionAndOrientation(self.DRONE_IDS[i], physicsClientId=self.CLIENT)
            #self.rpy[i] = p.getEulerFromQuaternion(self.quat[i])
            #self.vel[i], self.ang_v[i] = p.getBaseVelocity(self.DRONE_IDS[i], physicsClientId=self.CLIENT)
            pos, quat = p.getBasePositionAndOrientation(self.DRONE_IDS[i], physicsClientId=self.CLIENT)
            self.pos[i] = np.array(pos); self.quat[i] = np.array(quat)
            self.rpy[i] = p.getEulerFromQuaternion(self.quat[i])
            vel, ang_v = p.getBaseVelocity(self.DRONE_IDS[i], physicsClientId=self.CLIENT)
            self.vel[i] = np.array(vel); self.ang_v[i] = np.array(ang_v)
```
circumvents the problem?