learnsyslab/gym-pybullet-drones #4: Can I use reinforcement learning to control the drone navigate to the point goal ?

gmmhhh, 2020-09-29:


JacopoPan, 2020-09-29:
Have a look at class [`RLTakeoffAviary`](https://github.com/JacopoPan/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/RLTakeoffAviary.py) (used in script[`learn.py`](https://github.com/JacopoPan/gym-pybullet-drones/blob/master/examples/learn.py))

The current `_computeReward` is a step-wise function of the drone's z in the world frame

You can hard-code a goal position by making `_computeReward` return a (dense or sparse) reward computed as a function of the distance from the desired point

> Note that `_computeDone` terminates episodes outside the cube with edge=1, on top of the world frame origin (0,0,0)

> To fly outside this space, you'd need to re-implement both `_computeDone` and ` _clipAndNormalizeState`

closed 2020-09-30 by JacopoPan

chris-aeviator, 2021-01-13:
In my RL problem I'm introducing a design that has tilting motors, where I just mind about the thrust vectoring, not about collision of the propellers. I extended the actionspace for two additional values (n=6), made sure to set the upper and lower limits of the joints and I can confirm I can control them by setting a POSITION inside BaseAviary and see correct rotations in fly.py).

## 0 degree, 0 degree
[image]
## +30 degree (jointUpperLimit), -30 degree (jointLowerLimit)
[image]
## +30 degree ,+30 degree
[image]

I now wonder how `_clipAndNormalizeState` plays a role here. I cannot find any call to this function, not in this repo nor in openai/gym or all github :) .

 Do I append my two link positions (both can travel -30 to +30 degrees, set in RAD) that I want the singleAgent to optimize?

Would I modify  norm_and_clipped` inside HoverAviary (https://github.com/utiasDSL/gym-pybullet-drones/blob/f4d318679fccbc38dcf85dd9951149f8c61c5969/gym_pybullet_drones/envs/single_agent_rl/HoverAviary.py#L158)  and  `computeControl` https://github.com/utiasDSL/gym-pybullet-drones/blob/f4d318679fccbc38dcf85dd9951149f8c61c5969/gym_pybullet_drones/control/SimplePIDControl.py#L71 in SimplePIDController (which I'm using in baseAviary) to include the two angles of the motors?

JacopoPan, 2021-01-13:
@chris-aeviator 

I only briefly read this message (I'm not sure why it's in this issue?).
`_clipAndNormalizeState` is called in `_computeObs` of `BaseSingleAgentAviary.py` (or the multi agent class)

```
obs = self._clipAndNormalizeState(self._getDroneStateVector(0))
```

It's just an extra function to clip and normalized the state vector IF a certain RL problem requires it (or can benefit from it). 
It can be an identity function if you want to, you probably don't want to clip and normalize if you are using one of the PID controllers.

If you change the state vector, you would want to use `computeControl` instead of `computeControlFromState` in `BaseControl` (one is just a wrapper for the other to use fewer arguments).