learnsyslab/gym-pybullet-drones #5: Change of documentation format

ArciAndres, 2020-10-11:
This is not a feature request, but more like a suggestion. Until now the codes are nicely documented, but the format is not very conventional, as seen below.
 ```

    #### Initialize the environment ####################################################################
    ####################################################################################################
    #### Arguments #####################################################################################
    #### - drone_model (DroneModel)         desired drone type (associated to an .urdf file) ###########
    #### - num_drones (int)                 desired number of drones in the aviary #####################
    #### - neighbourhood_radius (float)     used to compute the drones' adjacency matrix, in meters ####
    #### - initial_xyzs ((3,1) array)       initial XYZ position of the drones #########################
    #### - initial_rpys ((3,1) array)       initial orientations of the drones (radians) ###############
    #### - physics (Physics)                desired implementation of physics/dynamics #################
    #### - freq (int)                       the frequency (Hz) at which the physics engine advances ####
    #### - aggregate_phy_steps (int)        number of physics updates within one call of .step() #######
    #### - gui (bool)                       whether to use PyBullet's GUI ##############################
    #### - record (bool)                    whether to save a video of the simulation ##################
    #### - obstacles (bool)                 whether to add obstacles to the simulation #################
    #### - user_debug_gui (bool)            whether to draw the drones' axes and the GUI sliders #######
    ####################################################################################################
 ```

I would suggest a more commonly used doctrings format for python (used in OpenAI gym repo), which has support in multiple IDEs in terms of snippets or giving extra information when hovering a function or class. This kind of formatting is well documented: https://www.python.org/dev/peps/pep-0257/

Awesome library! 

 ```

    '''
        Parameters
        ----------
        action : TYPE
            DESCRIPTION.

        Returns
        -------
        clipped_action : TYPE
            DESCRIPTION.
  '''

```

JacopoPan, 2020-10-11:
Thanks @ArciAndres, docstring is, in fact, in my TO-DO list for whenever I will go into a v1.x.x

While I develop, I prefer to use big chunky comment blocks because they make it easier on my eye to skim through the code

JacopoPan, 2020-10-12:
closing for now

closed 2020-10-12 by JacopoPan