learnsyslab/gym-pybullet-drones #7: Failed to make context current (eglError: 12288)

PandaNah, 2020-10-28:
I tried to look at the result of learning, but get an error.(`Failed to make context current (eglError: 12288)`)

Steps to reproduce the behavior:
learning commands: 
`conda activate env`
`python singleagent.py --env takeoff --algo ppo --pol mlp --input rpm`
test command: `python test_singleagent.py --file save-takeoff-ppo-mlp-rpm-10.28.2020_09.57.45`
This is result of test command:

pybullet build time: Oct  8 2020 00:10:46
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
argv[0]=--opengl2
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=3
argv[0] = --unused
argv[1] = --opengl2
argv[2] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
4

	visual 0x24 selected
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 1050/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 455.28
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
pthread_getconcurrency()=0
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/yoko/anaconda3/envs/UAV/lib/python3.7/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
Loaded EGL 1.5 after reload.
Failed to make context current (eglError: 12288)

JacopoPan, 2020-10-28:
Can you `git pull` and try again? I had uncommented this line that automatically loaded the egl plugin on Linux.

https://github.com/JacopoPan/gym-pybullet-drones/blob/b5f7cb5793afa7f9bb93d6550614fe87dfae790d/gym_pybullet_drones/envs/BaseAviary.py#L108

JacopoPan, 2020-10-28:
I'm closing the issue, please let me know if that was not the reason of your error.

closed 2020-10-28 by JacopoPan