learnsyslab/gym-pybullet-drones #10: Question on drone dynamics

thaipduong, 2020-11-25:
Hi Jacopo,
Thank you for working on this simulator. I have a question on the dynamics of the drones. In `gym-pybullet-drones/gym_pybullet_drones/envs/BaseAviary.py, line 851`:
`rpy = rpy + self.TIMESTEP * ang_v`

Please correct me if I'm wrong here. I think the angular velocity is generally not the same as roll, pitch and yaw rates so I wonder if this is a bug? To calculate the new rotation matrix after 1 TIMESTEP, I think it should be something like:
`next_rotation = matmul(rotation,expm(self.TIMESTEP * ang_v))`
And then the new roll, pitch, yaw can be calculated from this new rotation matrix.
Best.

JacopoPan, 2020-11-25:
You might be correct or those might be badly named variables.
I'll double check, but note that the function you are referring to, `_dynamics()` is currently unused, by default.

thaipduong, 2020-11-25:
I see. Thanks, Jacopo.

JacopoPan, 2020-11-28:
There was a mix-up in variable names as PyBullet's angular velocity vector was used as intermediate storage (with the intent of reducing code duplication) for the RPY rates (but not to update the pose) when choosing to call `_dynamics()` instead of `_physics`. It should be fixed in the next commit.

closed 2020-11-28 by JacopoPan