learnsyslab/gym-pybullet-drones #15: Questions about Downwash

yuchen-x, 2021-01-04:
Hi,

I am not a low-level control guy, according to my understanding, the _downwash() in BaseAviary.py is for automatically avoiding potential collisions with other drones. 

But, it seems only working for the drones flying at different height. For example, if I set the initial z=1 for both two drones in your examples/downwash.py, the two drones will collide or one of the drone suddenly going somewhere. 

I was wondering:

1) Is it possible to achieve the downwash when the drones flying at same height, and how?
2) How to set the downwash distance?
3) I modified the downwash.py to be with 3 drones (with initial z = [0.99, 0.999, 1.0] in the following video, I didn't get how the mechanism decides which drone to downwash, could you explain it to me? It seems not always the lower one to perform downwash, right?

https://user-images.githubusercontent.com/57424516/103563762-577a8500-4e8b-11eb-8c4d-7462d25b8a42.mp4

Thank you!

JacopoPan, 2021-01-05:
Hi @yuchen-x,

I think you misunderstood the intent of `_downwash()`. It is not a potential-based collision avoidance approach.
It is a physical phenomenon leading to [reduced lift in the turbulent air](https://www.grc.nasa.gov/WWW/k-12/airplane/downwash.html) generated by 2 drones crossing paths at different heights.

If you are interest in creating a repulsive force field around a drone, you could still exploit PyBullet's `applyExternalForce()` method and the information about the drones' relative distances. However, it wound't be very realistic nor within the scope of this simulation (for quadcopters).

yuchen-x, 2021-01-05:
I see，thank you very much for the quick reply and the explanation!

closed 2021-01-05 by yuchen-x