learnsyslab/gym-pybullet-drones #16: Ray dependency outdated

chris-aeviator, 2021-01-08:
**Describe the bug**
When installing as described and running `python examples/learn.py --rllib True` I'm faced with an `ImportError: libIlmImf-2_3.so.24: cannot open shared object file: No such file or directory` (Arch Linux). 

When looking at /usr/lib/ I can see that `pip install ray[rllib]` (or something different) installed libIlmImf-2_5.so.25 which seems to be coming from `openexr 2.5.3-6`

**To Reproduce**
Steps to reproduce the behavior:
`python examples/learn.py --rllib True`

**Expected behavior**
the correct version of 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Arch (up2date)
 - python 3.7
 - conda environment
 - Software and Packages Versions [e.g. Python 3.6, Rllib 0.8.4]

**Additional context**
Add any other context about the problem here.

chris-aeviator, 2021-01-08:
adding openxr 2.3 solves the issue for me but should be mentioned in the requirements. Also not installing tensorflow and using ``--rllib True`` results in an error

JacopoPan, 2021-01-08:
Thanks @chris-aeviator,

I don't see that error showing up (on macOS 11, python3.7), I'll look into it but, if 0.8.4 is the `ray` release you are using, I would definitely try to upgrade to a `ray-1.x.x`. 0.8.4 is from April last year and `ray`/`rllib` have been changing quite often and rapidly.

chris-aeviator, 2021-01-08:
@JacopoPan I'm using ray 1.1.0

JacopoPan, 2021-01-08:
Ok, I'll see if that's what's causing it

JacopoPan, 2021-01-08:
I upgraded to 1.1.0 but still don't see this error, was your `conda` environment a newly created one?

For reference, here is my `conda list` output
```
# packages in environment at /Users/jacopo/opt/anaconda3:
#
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0                    py37_0  
absl-py                   0.9.0                    pypi_0    pypi
aiohttp                   3.6.2                    pypi_0    pypi
aiohttp-cors              0.7.0                    pypi_0    pypi
aioredis                  1.3.1                    pypi_0    pypi
alabaster                 0.7.12                   py37_0  
anaconda                  2020.02                  py37_0  
anaconda-client           1.7.2                    py37_0  
anaconda-navigator        1.9.12                   py37_0  
anaconda-project          0.8.4                      py_0  
applaunchservices         0.2.1                      py_0  
appnope                   0.1.0                    py37_0  
appscript                 1.1.0            py37h1de35cc_0  
argh                      0.26.2                   py37_0  
asn1crypto                1.3.0                    py37_0  
astroid                   2.3.3                    py37_0  
astropy                   4.0              py37h1de35cc_0  
astunparse                1.6.3                    pypi_0    pypi
async-timeout             3.0.1                    pypi_0    pypi
atari-py                  0.2.6                    pypi_0    pypi
atomicwrites              1.3.0                    py37_1  
attrs                     19.3.0                     py_0  
autopep8                  1.4.4                      py_0  
babel                     2.8.0                      py_0  
backcall                  0.1.0                    py37_0  
backports                 1.0                        py_2  
backports.functools_lru_cache 1.6.1                      py_0  
backports.shutil_get_terminal_size 1.0.0                    py37_2  
backports.tempfile        1.0                        py_1  
backports.weakref         1.0.post1                  py_1  
beautifulsoup4            4.8.2                    py37_0  
bitarray                  1.2.1            py37h1de35cc_0  
bkcharts                  0.2                      py37_0  
blas                      1.0                         mkl  
bleach                    3.1.0                    py37_0  
blessings                 1.7                      pypi_0    pypi
blosc                     1.16.3               hd9629dc_0  
bokeh                     1.4.0                    py37_0  
boto                      2.49.0                   py37_0  
bottleneck                1.3.2            py37h776bbcc_0  
bzip2                     1.0.8                h1de35cc_0  
ca-certificates           2020.1.1                      0  
cachetools                4.1.1                    pypi_0    pypi
certifi                   2019.11.28               py37_0  
cffi                      1.14.0           py37hb5b8e2f_0  
chardet                   3.0.4                 py37_1003  
click                     7.0                      py37_0  
cloudpickle               1.3.0                      py_0  
clyent                    1.2.2                    py37_1  
colorama                  0.4.3                      py_0  
colorful                  0.5.4                    pypi_0    pypi
conda                     4.8.2                    py37_0  
conda-build               3.18.11                  py37_0  
conda-env                 2.6.0                         1  
conda-package-handling    1.6.0            py37h1de35cc_0  
conda-verify              3.4.2                      py_1  
contextlib2               0.6.0.post1                py_0  
cryptography              2.8              py37ha12b0ac_0  
curl                      7.68.0               ha441bb4_0  
cycler                    0.10.0                   py37_0  
cython                    0.29.15          py37h0a44026_0  
cytoolz                   0.10.1           py37h1de35cc_0  
dask                      2.11.0                     py_0  
[... 270 lines cut ...]
unixodbc                  2.3.7                h1de35cc_0  
urllib3                   1.25.8                   py37_0  
watchdog                  0.10.2           py37h1de35cc_0  
wcwidth                   0.1.8                      py_0  
webencodings              0.5.1                    py37_1  
werkzeug                  1.0.0                      py_0  
wheel                     0.34.2                   py37_0  
widgetsnbextension        3.5.1                    py37_0  
wrapt                     1.11.2           py37h1de35cc_0  
wurlitzer                 2.0.0                    py37_0  
xlrd                      1.2.0                    py37_0  
xlsxwriter                1.2.7                      py_0  
xlwings                   0.17.1                   py37_0  
xlwt                      1.3.0                    py37_0  
xmltodict                 0.12.0                     py_0  
xz                        5.2.4                h1de35cc_4  
yaml                      0.1.7                hc338f04_2  
yapf                      0.28.0                     py_0  
yarl                      1.5.0                    pypi_0    pypi
zeromq                    4.3.1                h0a44026_3  
zict                      1.0.0                      py_0  
zipp                      2.2.0                      py_0  
zlib                      1.2.11               h1de35cc_3  
zstd                      1.3.7                h5bba6e5_0  
```

chris-aeviator, 2021-01-08:
I think not the version of ray is the problem, the described libs that my installation complains about come from openexr - unfortunately I don't know how this openexr depedency comes in, not super familiar with how python pip packaging works.

My enevironment was freshly setup with conda for this project

JacopoPan, 2021-01-08:
I see, the only image processing library this repo makes use of is [`Pillow`](https://pillow.readthedocs.io/en/stable/installation.html#external-libraries) which, on Arch, was only tested with Python 3.8+

I'll leave this issue open for now.

closed 2021-02-27 by JacopoPan