learnsyslab/gym-pybullet-drones #19: Different result with explicit dynamics and Pybullet

shanki98, 2021-02-09:
I tried to build my own PID Controller (Despite the one given because I could not find documentation behind this). I created a single drone environment like:

INIT_XYZS = np.array([[0,0,0]]);
INIT_RPYS = np.array([[0,0,0]]);
env = CtrlAviary(drone_model=DroneModel.CF2X, # See DroneModel Enum class for other quadcopter models 
                 num_drones=1,                # Number of drones
                 initial_xyzs=INIT_XYZS,      # Initial XYZ position of the drone
                 neighbourhood_radius=np.inf, # Distance at which drones are considered neighbors, only used for multiple drones
                 physics=Physics.DYN,         # Choice of (PyBullet) physics implementation                                              
                 freq=500,                    # Stepping frequency of the simulation                    
                 aggregate_phy_steps=1,       # Number of physics updates within each call to BaseAviary.step()
                 gui=True,                    # Whether to display PyBullet's GUI, only use this for debbuging
                 record=False,                # Whether to save a .mp4 video (if gui=True) or .png frames (if gui=False) in gym-pybullet-drones/files/, see script /files/ffmpeg_png2mp4.sh for encoding
                 obstacles= False,            # Whether to add obstacles to the environment
                 user_debug_gui=False,        # Whether to use addUserDebugLine and addUserDebugParameter calls (it can slow down the GUI)
                 initial_rpys=np.array([0,0,0]).reshape(1,3))  
The PID Controller's architecture and design is similar to _Design of a trajectory tracking controller for a nanoquadcopter_ - Luis.C 2016. The controller works well in the model used in this paper and me (NL statespace with RK4 solver (Timestep = 0.002). (This is the reason I used 500 Hz as simulation frequency). The drones stabilizes almost stabilizes at a point in Pysics.DYN but Physics.PYB gives a bad result. Can you guide me to the changes that I would need to bring or to the right files I need to check?

JacopoPan, 2021-02-09:
Hi @shanki98,

try and use ```  before and after a block of code to properly display it in markdown

For example
```
x = 0
print(x)
```

`Physics.DYN` and `Physics.PYB` should work reasonably similarly (minus numerical approximation) at the default 240Hz stepping frequency. It might be possible that they give you different results at much lower or higher frequencies but I don't think it should be the case with 500Hz.

Assuming that your top level script is properly set up, all the logic you are concerned to is run by the step in `BaseAviary`
https://github.com/utiasDSL/gym-pybullet-drones/blob/beb4fff10421f86da4d02addc194c833c7764d48/gym_pybullet_drones/envs/BaseAviary.py#L269

My suggestion is to make sure that your control and simulation frequencies match/are properly aligned. If you are willing to share your code, I will have a look at it.

shanki98, 2021-02-10:
Thanks for getting back to me I can share my code. 
'''
import numpy as np
import matplotlib.pyplot as plt
import math
import time
import IPython
from Reference_CF import CFreference
from gym_pybullet_drones.envs.BaseAviary import DroneModel, Physics, BaseAviary
from gym_pybullet_drones.envs.CtrlAviary import CtrlAviary
from gym_pybullet_drones.control.BaseControl import BaseControl
from utils import *
##### To remove any existing registrations of environments to avoid re-registry
import gym
env_dict = gym.envs.registration.registry.env_specs.copy()
for i in env_dict:
    if 'ctrl-aviary-v0' or 'takeoff-aviary-v0' or 'flock-aviary-v0' in i:
        print("Remove {} from registry".format(i))
        del gym.envs.registration.registry.env_specs[i]
INIT_XYZS = np.array([[0,0,0]]);
INIT_RPYS = np.array([[0,0,0]]);
env = CtrlAviary(drone_model=DroneModel.CF2X, # See DroneModel Enum class for other quadcopter models 
                 num_drones=1,                # Number of drones
                 initial_xyzs=INIT_XYZS,      # Initial XYZ position of the drone
                 neighbourhood_radius=np.inf, # Distance at which drones are considered neighbors, only used for multiple drones
                 physics=Physics.PYB,         # Choice of (PyBullet) physics implementation                                              
                 freq=500,                    # Stepping frequency of the simulation                    
                 aggregate_phy_steps=1,       # Number of physics updates within each call to BaseAviary.step()
                 gui=True,                    # Whether to display PyBullet's GUI, only use this for debbuging
                 record=False,                # Whether to save a .mp4 video (if gui=True) or .png frames (if gui=False) in gym-pybullet-drones/files/, see script /files/ffmpeg_png2mp4.sh for encoding
                 obstacles= False,            # Whether to add obstacles to the environment
                 user_debug_gui=False,        # Whether to use addUserDebugLine and addUserDebugParameter calls (it can slow down the GUI)
                 initial_rpys=np.array([0,0,0]).reshape(1,3))       
parameters = BaseControl(env);
class CFParameters:
    def __init__(self):
        self.m = 0.027;
        self.L = 0.039730;
        self.KF = parameters.KF;
        self.KM = parameters.KM;
        self.MAX_THRUST = parameters.MAX_THRUST; 
        self.MAX_XY_TORQUE = parameters.MAX_XY_TORQUE; 
        self.MAX_Z_TORQUE = parameters.MAX_Z_TORQUE;
        self.I = np.diag([0.00001395,0.00001436,0.00002173]);
        self.Pos_ctrl_HZ = 100;
        self.Att_ctrl_HZ = 250;
        self.Rate_ctrl_HZ = 500;
        self.MIN_PWM = 20000; 
        self.MAX_PWM = 65535;
        self.CF_vel = 1.2;#m/s
g = 9.8;
cf2x = CFParameters();
Ts_Pos = 1/cf2x.Pos_ctrl_HZ;
Ts_att = 1/cf2x.Att_ctrl_HZ;
Ts_rate = 1/cf2x.Rate_ctrl_HZ;
Duration_sec = 15;
#Duration_sec = 40;
T_out = np.arange(0,Duration_sec,Ts_Pos);
T_rate = np.arange(0,Duration_sec,Ts_rate);
T_rpref = np.arange(0,Duration_sec,15/4500);
#T_rpref = np.arange(0,Duration_sec,Duration_sec/12000);
#Traj = CFreference(Duration_sec=40);
Traj = CFreference(Duration_sec=15);
Time = 0.0;
Traj.Linear(Time);
#Traj.Helical(Time);
#Traj.Circular(Time,Duration_sec=40);
k_p_x =1.59;
k_i_x =5.73;
k_p_y =-1.59;
[... 205 lines cut ...]
        else:
            z = obs['0']['state'][2];
            z_dot = obs['0']['state'][12];
            psi = obs['0']['state'][9];
            p = obs['0']['state'][13];
            q = obs['0']['state'][14];
            r = obs['0']['state'][15];
            Omega,integral_alt_e = Altitudecontrol(Traj,z,z_dot,idx,integral_alt_e,Ts_rate);
            RPM_1,RPM_2,RPM_3,RPM_4,integral_psir_e = Ratecontrol(des_p,des_q,des_r,Omega,p,q,r,integral_psir_e,Ts_rate);
            Input = dict({'0':[RPM_1,RPM_2,RPM_3,RPM_4]});
            obs, reward, done, info = env.step(Input);
            xs.append(obs['0']['state'][0]);
            ys.append(obs['0']['state'][1]);
            zs.append(obs['0']['state'][2]);
            phis.append(obs['0']['state'][7]);
            thetas.append(obs['0']['state'][8]);
            psis.append(obs['0']['state'][9]);
            env._showDroneLocalAxes(0)
            env.render()
            if done: obs = env.reset()
            #### Sync the simulation ###########################################################################
            sync(_,START, env.TIMESTEP)
            Time = Time + env.TIMESTEP;
env.close()
'''

JacopoPan, 2021-02-10:
I think I see what might lead you to different results. In the current commit, `obs['0']['state'][13:16]` contains the component of the angular velocity, not the turn rates, for `Physics.PYB` and 0s for `Physics.DYN`

You could try to replace the occurrences of these lines
```
p = obs['0']['state'][13];
q = obs['0']['state'][14];
r = obs['0']['state'][15];
```
with just
```
p = 0
q = 0
r = 0
```
or something like (but you'll need to store the previous rpy's)
```
p = (obs['0']['state'][7]-last_phi)/timestep
q = (obs['0']['state'][8]-last_theta)/timestep
r = (obs['0']['state'][9]-last_psi)/timestep
```
If nothing else, the behaviour should be the same then.

shanki98, 2021-02-12:
Hey Jacopo,
thanks for looking into my code. I really appreciate it. I corrected the p,q,r though I'm getting an oscillatory response. It is better than the previous one. Can you tell me if Wx, Wy, Wz are angular velocities measured in terms of body-frame or inertial frame?

JacopoPan, 2021-02-12:
I would try to tune the PID coefficient first, then. The angular velocity values are only stored/available when using `Physics.PYB`, not `Physics.DYN`. They are those returned by PyBullet's [`getBaseVelocity` method](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.4vxw9j7piyjd).

shanki98, 2021-02-15:
Finally, It worked after retuning the PID coefficients a little bit. But I should say the stepSimulation of pybullet did give a different response than the explicit dynamics before I corrected it. (I checked this by giving the same input and observing the response of the system in both cases for a single step).

For p,q,r, I did : 
```
 wx,wy,wz = env.ang_v[0,:];
 p,q,r =  np.dot(rotation.transpose(),np.array([wx,wy,wz]));
```
This gave me the p,q,r in the inertial frame. Then after just retuning the PID coefficients a liittle bit, I got convergence!

Thanks a lot for your help !!

JacopoPan, 2021-02-15:
I'm glad you managed to have your controller to work. You should keep in mind that `env.ang_v[0,:]` is only written for `Physics.PYB`. I believe that's were your implementation differs (minus, possibly, numerical approximation in near zero values).

closed 2021-02-15 by JacopoPan