learnsyslab/gym-pybullet-drones #21: Action explain

azamkamranian, 2021-02-10:
Can someone explain for me what is  each element of action? I didnt find any wiki for this project!

JacopoPan, 2021-02-10:
Hi @azamkamranian,

there is a note on that in the [README](https://github.com/utiasDSL/gym-pybullet-drones#action-spaces-examples). The idea is that one can re-define methods `_actionSpace()` and `_preprocessAction()` for these to be arbitrary implementations. In class [`CtrlAviary`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/CtrlAviary.py), an action for `N` drones is a dictionary in the form
```
{"0": [RPM_P0, RPM_P1, RPM_P2, RPM_P3], "1": [RPM_P0, RPM_P1, RPM_P2, RPM_P3], .., "N": [RPM_P0, RPM_P1, RPM_P2, RPM_P3]}
```

Class [`BaseSingleAgentAviary`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/single_agent_rl/BaseSingleAgentAviary.py) contains a few different implementations—ranging from a single value for the RPMs applied to each propeller to a desired velocity vector (targeted via PID control)—listed in `Enum` [`ActionType`](https://github.com/utiasDSL/gym-pybullet-drones/blob/beb4fff10421f86da4d02addc194c833c7764d48/gym_pybullet_drones/envs/single_agent_rl/BaseSingleAgentAviary.py#L14)

azamkamranian, 2021-02-10:
RPM : round per minute , now I got this action concept.

JacopoPan, 2021-02-10:
Yes, you can think of it a linear transformation away from PWM (and maybe it's a bit more intuitive).

closed 2021-02-10 by JacopoPan