learnsyslab/gym-pybullet-drones #22: has anyone successfully trained multi agent problem : flockaviary?

MuhammadFadhilArkan, 2021-02-15:
First of all, thankyou for your work on this simulator.

after training flockaviary environment using multiagent.py for 200 iteration
[image]

then testing it using test_multiagent.py, the result is not as expected 
[image]

as you can see, z position is above 100 even though from _computereward function we can see that the goal is to make multiple drones hover at z position =1 
[image]

after looking at progress.csv, the reward is already steady (around -29) , which i assume that increasing the number of iteration will not affect the performance.

is this a problem with environment (flockaviary.py) or the algorithm (multiagent.py) ?

JacopoPan, 2021-02-15:
> is this a problem with environment (flockaviary.py) or the algorithm (multiagent.py) 

(there's something to say about action/observation types and the policy architectures but) the short answer is *both*: the reward function in `flockaviary.py` is the one you mention, not one necessarily intended for flocking, and `multiagent.py` is still the least mature piece of code in this repo (e.g., it needs work to extend to more than 2 drones)

you can use `test_multiagent.py` to have a look at what the trained policies are (and maybe infer where it is going wrong)

JacopoPan, 2021-02-23:
I realized that `test_multiagent.py` was not working properly because the input to `compute_single_action()` was not in the correct order. The default execution of script `multiagent.py` (a 1D leader-follower example) can now be trained and tested in ~10'
[image]

closed 2021-04-09 by JacopoPan