learnsyslab/gym-pybullet-drones #24: Question: Is there a way to control the drones by 1D thrust and 3D torques on the center of mass?

thaipduong, 2021-02-22:
The control seems to be rpm of each rotors in fly.py. Is there a way to control the drones by 1D thrust and 3D torques on the center of mass?
Thank you. Appreciate your help.

JacopoPan, 2021-02-22:
Hi @thaipduong,

I wrote this alternative aviary, internally using non-negative least squares to compute RPMs from input thrust and torques:
https://github.com/utiasDSL/gym-pybullet-drones/blob/962631269fcfe5ac2032a394248ec18aa6b72bf2/gym_pybullet_drones/envs/DynAviary.py#L7
https://github.com/utiasDSL/gym-pybullet-drones/blob/962631269fcfe5ac2032a394248ec18aa6b72bf2/gym_pybullet_drones/utils/utils.py#L58
I never used it extensively, though. Could you try it and see if it does the job you are looking for?

thaipduong, 2021-02-22:
Thank you, @JacopoPan. I'll give it a try :).

closed 2021-02-22 by thaipduong