learnsyslab/gym-pybullet-drones #28: Cannot connect to X server

alexvaca0, 2021-04-08:
**Describe the bug**
When running in either a ubuntu18 or in Google Colab, I consistently receive the following error when using learn.py: Cannot connect to X server.
**To Reproduce**
Steps to reproduce the behavior:
1. Run script examples/learn.py in Google colab.

**Expected behavior**
I would expect to be able to see the videos of the drone flying, but I'm not able to.

JacopoPan, 2021-04-08:
Hi @alexvaca0 

I suppose this happens when creating an env with GUI at this line?
https://github.com/utiasDSL/gym-pybullet-drones/blob/333ec90219bd39fe69aad53734a21f27faf2449b/examples/learn.py#L78

I don't think you can start PyBullet's GUI from Colab, you should be able to avoid the error by setting `gui=False`

If you want to implement rendering in Colab you can look at [this example](https://colab.research.google.com/drive/1u6j7JOqM05vUUjpVp5VNk0pd8q-vqGlx?authuser=1#scrollTo=7tbOVtFp1_5K
) exploiting `p.getCameraImage` and `IPython.display`

Camera images are actually saved in `BaseAviary` at this line
https://github.com/utiasDSL/gym-pybullet-drones/blob/333ec90219bd39fe69aad53734a21f27faf2449b/gym_pybullet_drones/envs/BaseAviary.py#L301

alexvaca0, 2021-04-08:
Hi @JacopoPan , thanks a lot for the quick response, you're being really helpful here :) 

Yes, I'm leaving gui=True at that line, my intention was to watch the video dynamically as it's rendered from the env. Okay so first thing I'll do is set that argument to False. 
I'm checking your example and it seems understandable, therefore instead of using the PyBullet's GUI, what I'd need to do is load the environment and my agent in the notebook, and then run some episodes, saving the images in the environment object, and accessing them to save all the frames with which I can make the video, right?

JacopoPan, 2021-04-08:
Happy to (try to) help @alexvaca0,

admittedly, the current implementation is not intended for graphical display/rendering with Colab.

What I do is `BaseAviary` is using `p.getCameraImage` to save frames when `gui == False` and `record == True`
These images are taken at `self.CAPTURE_FREQ` and saved under `self.IMG_PATH`
https://github.com/utiasDSL/gym-pybullet-drones/blob/333ec90219bd39fe69aad53734a21f27faf2449b/gym_pybullet_drones/envs/BaseAviary.py#L310

What I'm suggesting you could do is to append these frames in an array, make a video, and show it in Colab as in the example linked above (with the caveat that you are rendering it after the experiment, not during it)
```
print("creating animated png, please about 5 seconds")
%time write_apng("example6.png", frames, delay=100)
%time Image(filename="example6.png")
```

closed 2021-04-12 by JacopoPan