learnsyslab/gym-pybullet-drones #30: accessing camera for RL

JGittles, 2021-04-10:
Hi. I was trying to figure out how I can access the depth camera stream for some RL applications.

is there any documentation on how I can access this information? I noticed older iterations had a __dev.py folder with a list of features but the newest master doesn't seem to contain this.

Thanks!

JacopoPan, 2021-04-10:
Hey @JGittles,

you can simply use [class `VisionAviary`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/VisionAviary.py) (look at its observation space)
https://github.com/utiasDSL/gym-pybullet-drones/blob/333ec90219bd39fe69aad53734a21f27faf2449b/gym_pybullet_drones/envs/VisionAviary.py#L97

The `obs` returned by `obs, reward, done, info = env.step(..)`, should include a `Box(H,W)` for floats, representing the depth information, under `obs['drone_id']['dep']`

Some of the options with which those frames are captured (frequency, resolution)  are specified in `BaseAviary`
https://github.com/utiasDSL/gym-pybullet-drones/blob/333ec90219bd39fe69aad53734a21f27faf2449b/gym_pybullet_drones/envs/BaseAviary.py#L153

JGittles, 2021-04-10:
Awesome thanks! I'll give it a look.

closed 2021-04-11 by JGittles