learnsyslab/gym-pybullet-drones #34: Simple question

Suvian-wy, 2021-04-20:
Firstly, I'm sorry to disturb you!  I want to know whether the model is an accurate model of Crazyfile2. I mean I don't know wherher the aerodynamics is considered. If I implement a low-level controller and it perform will in gym-pybullet-drones. Will the controller perform well in a hardware Crazyfile2?  Because I want to transplant a RL attitude controller from simulation environment to real platforma. I don't konw if this simulation can do it. Thanks.

JacopoPan, 2021-04-20:
Hi @Suvian-wy,

the physical property (mass, inertia) of the UAV models that are in the URDF files, e.g. [`cf2x.urdf`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/assets/cf2x.urdf), are taken from the system identification work at ETH and Polytechnique Montreal of
- Julian Forster ([*System Identification of the Crazyflie 2.0 Nano Quadrocopter*](http://mikehamer.info/assets/papers/Crazyflie%20Modelling.pdf))
- and Carlos Luis ([*Design of a Trajectory Tracking Controller for a Nanoquadcopter*](https://arxiv.org/pdf/1608.05786.pdf))

I would keep in mind a couple of things: _(i)_ adding decks, reflective markers, etc. certainly affects the mass of a CF2.x, _(ii)_ the default action space of this gym contains propellers' RPMs (and thrust is assumed perfectly proportional to their squared values). On a practical on-board implementation, you'd probably want to set PWMs: a brief quantitative investigation of the relation between PWM/RPM/thrust can be found [here](https://wiki.bitcraze.io/misc:investigations:thrust).

Suvian-wy, 2021-04-21:
OK! Thanks!

closed 2021-04-21 by Suvian-wy