learnsyslab/gym-pybullet-drones #36: Occupancy grid

kevin-thankyou-lin, 2021-04-25:
Thanks for the great project and codebase! Is there a way to get a 3D occupancy grid with which I can do obstacle avoidance trajectory generation?

JacopoPan, 2021-04-26:
This is not something currently implemented but I could give it some thought, can you give me a small example of the scenario/output you would like to see? Should the occupancy grid be part of the `obs` (i.e. do you need it for RL) or could it be an additional entry in `info` (i.e. you are using the environment simply as a simulation?

kevin-thankyou-lin, 2021-04-27:
Gotcha, I was looking more for something like:

Airsim's Voxel grid construction (https://github.com/microsoft/AirSim/pull/3209)

to get the 'ground truth' 3D state of the world. I'd use it as an additional entry in `info` - specifically, I'd want to use it for ground truth optimal planning.

I think we could use something like this voxelization tool for the .obj files for the sphere, samurai, etc URDFs (http://www.open3d.org/docs/latest/tutorial/Advanced/voxelization.html).

That said, I think I'm good without an occupancy grid at this stage, but thank you for helping!

JacopoPan, 2021-04-27:
I should give it some thought. Note that collisions are managed by PyBullet (see the [collision queries api](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.hrpbzrwzgy1p)) using the collision shapes specified in the .urdf file of each drone/object (these could be simple geometries or .dae/.obj meshes). The ground truth position/orientation of any object in the sim can be retrieved with a call to `p.getLinkState()`, then (knowing the collision shape/mesh of each object) one could use Open3D's [voxelization](http://www.open3d.org/docs/latest/tutorial/Advanced/voxelization.html). I have to look into how to use basic geometries and multiple meshes in a single Voxel grid but I suspect it could result in a significative computational overhead.

kevin-thankyou-lin, 2021-04-27:
Sounds good! Yeah, it'd probably result in significant computational overhead.

closed 2021-05-20 by JacopoPan