learnsyslab/gym-pybullet-drones #41: GUI rendering does not work properly

GPrathap, 2021-05-03:
**Describe the bug**
GUI rendering does not work properly 

However, backend looks fine

jobs@gdnpc:~/sim/gym-pybullet-drones/examples$ python3 fly.py
pybullet build time: Apr 25 2021 07:44:46
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce RTX 2080 Ti/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 460.73.01
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 460.73.01
Vendor = NVIDIA Corporation
Renderer = GeForce RTX 2080 Ti/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/geesara/.local/lib/python3.6/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
ven = NVIDIA Corporation
ven = NVIDIA Corporation

...
**To Reproduce**
Just trying out python3 fly.py example 

**Expected behavior**
Not clear, why GUI is working properly 

**Screenshots**
This is the kind of output I am getting 

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 18, 
 - Software Python 3.6

JacopoPan, 2021-05-04:
I wish I had a 2080Ti to replicate the issue but it really seems more likely an OpenGL/graphics problem.
Do you have OpenGL installed? Can you try this minimal PyBullet example and see if the problem persists?
If so, try to replace line `p.connect(p.GUI)` with `p.connect(pybullet.GUI, options="--opengl2")`
If nothing changes, let me know which version of PyBullet you are using.
```
import pybullet as p
import pybullet_data
from time import sleep

def main():
    PYB_CLIENT = p.connect(p.GUI)
    p.setGravity(0, 0, -9.8, physicsClientId=PYB_CLIENT)
    p.setRealTimeSimulation(0, physicsClientId=PYB_CLIENT)
    p.setTimeStep(1/240, physicsClientId=PYB_CLIENT)
    p.setAdditionalSearchPath(pybullet_data.getDataPath(), physicsClientId=PYB_CLIENT)
    p.loadURDF("plane.urdf", physicsClientId=PYB_CLIENT)
    p.loadURDF("duck_vhacd.urdf", [0, 0, 1], physicsClientId=PYB_CLIENT)
    for _ in range(240*10):
        p.stepSimulation(physicsClientId=PYB_CLIENT)
        sleep(1/(240*10))

if __name__ == "__main__":
    main()
```

GPrathap, 2021-05-04:
Thank you for the quick reply, Yep this example works fine, 

[image]

Yes, I have OpenGL 

```
jobs@gdnpc:~/sim/gym-pybullet-drones/files/videos$ glxinfo | grep "OpenGL version"
OpenGL version string: 4.6.0 NVIDIA 460.73.01

jobs@gdnpc:~/sim/gym-pybullet-drones/files/videos$ glxinfo | grep 'version'
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 460.73.01
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 460.73.01
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 460.73.01
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
 GL_EXT_shader_group_vote, GL_EXT_shader_implicit_conversions, 
```

How can I modify the GUI client here https://github.com/utiasDSL/gym-pybullet-drones/blob/master/examples/fly.py#L123 

I see it is  constructed here: https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/BaseAviary.py#L180 yet not clear how to override this

GPrathap, 2021-05-04:
I modified (https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/BaseAviary.py#L180 replace `p.connect(p.GUI)` with `p.connect(p.GUI, options="--opengl2")` ). However, still  having the same problem (: 

jobs@gdnpc:~/sim/gym-pybullet-drones$ /usr/bin/python3 /home/jobs/sim/gym-pybullet-drones/examples/fly.py
pybullet build time: Apr 25 2021 07:44:46
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
argv[0]=--opengl2
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=3
argv[0] = --unused
**argv[1] = --opengl2**
argv[2] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
4

        visual 0x24 selected
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce RTX 2080 Ti/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 460.73.01
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
pthread_getconcurrency()=0
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/geesara/.local/lib/python3.6/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))

GPrathap, 2021-05-04:
I tried on inside a docker image, it works fine, I guess something wrong with my graphs, if you have time will have look. If not its fine, I will continue with docker image. Thanks for the help

JacopoPan, 2021-05-04:
I am not sure, the following 2 snippets are where all the PyBullet GUI options are set, but they essentially do the same things that the minimal example above does too. You could try to comment out as many of the `p.methodName(..)` lines to see if any one of the them is the root of your issue.

https://github.com/utiasDSL/gym-pybullet-drones/blob/7bec934d0dd4dec5135a4e5d8b6d3809043d5856/gym_pybullet_drones/envs/BaseAviary.py#L178-L197

https://github.com/utiasDSL/gym-pybullet-drones/blob/7bec934d0dd4dec5135a4e5d8b6d3809043d5856/gym_pybullet_drones/envs/BaseAviary.py#L483-L499

GPrathap, 2021-05-05:
I tried one of the AirSim simulators and it also crashes immideatly after starts,  I guess something wrong with graphics drivers

closed 2021-05-05 by JacopoPan