learnsyslab/gym-pybullet-drones #42: Possibility of software-in-the-loop (SIL/SITL) testing

dominiknatter, 2021-05-06:
Thanks for this cool project - I'm sure it will push research a step forward!

I came across it while working with Crazyflies and was excited that you use the Crazyflie as your default quadrotor. Please let me know if I understand the details correctly: you use only the dynamical properties of the Crazyflie for this default quadrotor, but the project does not involve the Crazyflie firmware at all, right? If that is correct, software-in-the-loop (SIL/SITL) testing for the Crazyflie is something your approach has not tackled (yet).

Thanks for your help! Best,
Dominik

JacopoPan, 2021-05-06:
Yes @dominiknatter, you are correct.

The point here is the simulation, reinforcement learning, and control of dynamical systems and I choose the CF because it's a platform I like, use, and with a lot of system identification work available.

There are similarities between the handwritten PID controllers and the [CF's firmware control modules](https://github.com/bitcraze/crazyflie-firmware/tree/master/src/modules/src) but, ultimately, one should re-define action and observations spaces depending on the type of controller they want to implement and port the trained agent to the firmware. (Of course it's something that has been at the back of my mind from the incipit but this public repo does not include any work in that direction.)

dominiknatter, 2021-05-07:
Thanks for the clarifications @JacopoPan!

closed 2021-05-07 by dominiknatter

soldierofhell, 2021-06-29:
> ultimately, one should re-define action and observations spaces depending on the type of controller they want to implement and port the trained agent to the C firmware. (Of course it's something that has been at the back of my mind from the incipit but this public repo does not include any work in that direction.)

Hi @JacopoPan ,
I've got few questions regading "deploying" the learned policies into flight controller if you mind:
* This Crazyflie MCU (Cortex-M4) is not the most performant one on the market, don't you think?
* AFAIK Crazyflie firmware is in C, this could be a problem if the code is in C++?
* Are currently used RL frameworks (stable_baselines3, rllib) capable of deploying stripped down code suitable for MCU deployment? I thought the smoothed path is through TFLite and thought about tf_agents, although there're still some unsupported TFLite ops right now. I don't know about PyTorch, maybe through onnx and TVM?

Regards,