learnsyslab/gym-pybullet-drones #45: GUI does not work properly

stavrako, 2021-05-18:
**Describe the bug**
Same as #41. 
Installation steps are applied properly.

Trying to run fly.py.

stavrako@stavrako:~/gym-pybullet-drones/examples$ python3 fly.py
pybullet build time: May 14 2021 14:07:06
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce MX110/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 460.73.01
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 460.73.01
Vendor = NVIDIA Corporation
Renderer = GeForce MX110/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/stavrako/.local/lib/python3.6/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
ven = NVIDIA Corporation
ven = NVIDIA Corporation

[INFO] BaseAviary.render() ——— it 0005 ——— wall-clock time 1.5s, simulation time 0.0s@240Hz (0.01x)
[INFO] BaseAviary.render() ——— drone 0 ——— x +00.00, y +00.00, z +00.10 ——— velocity +00.00, +00.00, +00.00 ——— roll +00.00, pitch -00.00, yaw +00.00 ——— angular velocity +0.0000, +0.0000, +0.0000 ——— 
[INFO] BaseAviary.render() ——— drone 1 ——— x -00.26, y -00.15, z +00.15 ——— velocity +00.00, +00.00, +00.00 ——— roll -00.00, pitch -00.00, yaw +30.00 ——— angular velocity +0.0000, +0.0000, +0.0000 ——— 
[INFO] BaseAviary.render() ——— drone 2 ——— x -00.26, y -00.45, z +00.20 ——— velocity +00.00, +00.00, +00.00 ——— roll -00.00, pitch -00.00, yaw +60.00 ——— angular velocity +0.0000, +0.0000, +0.0000 ——— 

...

numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

[image]

**To Reproduce**
Steps to reproduce the behavior:
Running fly.py

**Expected behavior**

1. 

I run a minimal example that has been referred in #41 
`import pybullet as p
import pybullet_data
from time import sleep

def main():
    PYB_CLIENT = p.connect(p.GUI)
    p.setGravity(0, 0, -9.8, physicsClientId=PYB_CLIENT)
    p.setRealTimeSimulation(0, physicsClientId=PYB_CLIENT)
    p.setTimeStep(1/240, physicsClientId=PYB_CLIENT)
    p.setAdditionalSearchPath(pybullet_data.getDataPath(), physicsClientId=PYB_CLIENT)
    p.loadURDF("plane.urdf", physicsClientId=PYB_CLIENT)
    p.loadURDF("duck_vhacd.urdf", [0, 0, 1], physicsClientId=PYB_CLIENT)
    for _ in range(240*10):
        p.stepSimulation(physicsClientId=PYB_CLIENT)
        sleep(1/(240*10))

if __name__ == "__main__":
    main()`

[image]

2. "replace p.connect(p.GUI) with p.connect(p.GUI, options="--opengl2")" didn't work as well.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Software  Python 3.6
 - 
[image]

**Additional context**
I have tried Windows and it works fine there.

JacopoPan, 2021-05-18:
Thanks @stavrako,

I'd suspect this to be an issue arising directly from PyBullet or OpenGL when using Ubuntu and NVIDIA GPUs.
I normally work on macOS (and I'm glad Windows works fine too) but I will look into this when I can.

lampidis, 2021-05-20:
I experienced the same problem as well

JacopoPan, 2021-05-20:
Ok, I'll look into this today. @stavrako could you add the terminal output when running the minimal example I gave in https://github.com/utiasDSL/gym-pybullet-drones/issues/41?

JacopoPan, 2021-05-20:
umm, I have a different card and version of NVIDIA driver but I don't seem to be able to replicate the issue
```
(base) jacopo@jacopo-ThinkPad-P52:~/Desktop/gym-pybullet-drones$ python examples/fly.py 
pybullet build time: May 20 2021 11:07:34
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=Quadro P2000/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 450.119.03
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 450.119.03
Vendor = NVIDIA Corporation
Renderer = Quadro P2000/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/jacopo/anaconda3/lib/python3.7/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
ven = NVIDIA Corporation
ven = NVIDIA Corporation
```
[image]

JacopoPan, 2021-05-20:
@stavrako you have the same driver as @GPrathap in https://github.com/utiasDSL/gym-pybullet-drones/issues/41 so I think the problem might be related to that.

However, it is curious about why the minimal example does not raise the same problem.

Can you try if commenting these 2 lines changes anything?

https://github.com/utiasDSL/gym-pybullet-drones/blob/7bec934d0dd4dec5135a4e5d8b6d3809043d5856/gym_pybullet_drones/envs/BaseAviary.py#L9-L10

JacopoPan, 2021-05-20:
You can try to comment out other PyBullet GUI-related calls like `p.configureDebugVisualizer()` and `p.resetDebugVisualizerCamera()` but I don't see any other way in which the simple example from above and `BaseAviary` actually differ. If you find out, please let me know.
I wish I could help but every works as expected on Ubuntu 18 and with a Quadro P2000 for me.

stavrako, 2021-05-20:
I understand.
Thank you for your time.

lampidis, 2021-05-20:
I have a Nvidia 1660 super so I think it has something to do with the Nvidia drivers for ubuntu. On windows (same graphics card) it is working as expected.

JacopoPan, 2021-05-20:
Thanks for confirming that @lampidis.
If you happen find a way to replicate the problem in the minimal example, I'd try to fix it (or at least understand the cause).

closed 2021-05-24 by JacopoPan