learnsyslab/gym-pybullet-drones #47: single agent difference

azamkamranian, 2021-05-26:
Hi. Could someone explain what is difference between agents "takeoff-aviary-v0", "hover-aviary-v0", and "flythrugate-aviary-v0"?

JacopoPan, 2021-05-27:
Hi @azamkamranian,

There are essentially 3 layers of implementation:
- `BaseAviary` is a class that implements all is needed to simulate one or more quadcopter (but not RL interfaces)
- Subclasses `BaseSingleAgentAviary` and `BaseMultiagentAviary` extend it with Gym-style action and observation spaces
- Finally, subsubclasses like `HoverAviary` implement the task-specific methods of each Gym (reward, done, info) and other tweaks like normalization and clipping of the observation, obstacles, etc.

azamkamranian, 2021-05-27:
@JacopoPan thank you for the explanation. yes, I got it.

azamkamranian, 2021-05-27:
@JacopoPan I know there is so much work but I think a wiki really helps. not only for envs, but a little explaining about the quadcopter physics and ctrl stuff. It really helps to understand functions and ctrl methods for those who are new to drones.

JacopoPan, 2021-05-27:
I am typically against over complicating documentations but I admit that this repo ballooned a bit w.r.t. to its initial README.
I'm currently working on a new repo with a couple of simple single agent control tasks that should make it easier to approach RL with a controls background and vice versa. I'll revisit this one once the other is released.

azamkamranian, 2021-05-27:
That is so great.  I want to use new envs for implementing RL algorithms. So, I am looking forward to the new rep too.

xiaomaohouhou, 2022-05-15:
gym.error.UnregisteredEnv: No registered env with id: takeoff-aviary-v0

The terminal reminds me of this error. Can you help me? Thank you!

JacopoPan, 2022-05-15:
Hi @xiaomaohouhou 

did you `pip3 install -e .` in the main folder? The registration of the environment is in `gym-pybullet-drones/gym_pybullet_drones/__init__.py`.

closed 2022-08-03 by JacopoPan