learnsyslab/gym-pybullet-drones #48: voxel_grid

p00uya, 2021-05-31:
**Describe the bug**
i wanna to test script as mentioned at the end of [link](https://github.com/microsoft/AirSim/blob/6d20663e0818448613be0332009b02bd6341589e/docs/voxel_grid.md)

when i run this script :

```
import airsim

c = airsim.VehicleClient()
center = airsim.Vector3r(0, 0, 0)
output_path = os.path.join(os.getcwd(), "map.binvox")
c.simCreateVoxelGrid(center, 100, 100, 100, 0.5, output_path)
```

**python said: VehicleClient has no attribute simCreateVoxelGrid**
did anything need to write before or after?

**Desktop (please complete the following information):**
 - OS: win10
 - Software and Packages Versions  Python 3.9 & 3.8

JacopoPan, 2021-05-31:
Hi @p00uya,

I think you meant to open an issue on https://github.com/microsoft/AirSim instead? 😅

p00uya, 2021-05-31:
Hi yeah

closed 2021-05-31 by JacopoPan

pouyakhani1374, 2021-05-31:
🙈😄👍