learnsyslab/gym-pybullet-drones #49: How to move drone instantaneously?

CoolCoder54323, 2021-05-31:
Is there a way to set the drone's position, RPY, and velocities instantaneously without having to use physics?

Also, I'm having a hard time learning about the features of this library. Is there any documentation besides the readme?

JacopoPan, 2021-05-31:
I'd refer you to [PyBullet's Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3) for resetting links (and joints). You can try something similar to:
```
PYB_CLIENT = env.getPyBulletClient()
DRONE_IDS = env.getDroneIds()
p.resetBasePositionAndOrientation(DRONE_IDS[nth_drone],
                                  [x, y, z],
                                  p.getQuaternionFromEuler([r, p, y]),
                                  physicsClientId= PYB_CLIENT
                                  )
p.resetBaseVelocity(DRONE_IDS[nth_drone],
                    [vx, vy, vz],
                    physicsClientId= PYB_CLIENT
                    )
```
At this time, the repo is documented:
- by the main README.md
- through NumPy-style doc strings in all source code files
- and in this pre-print ([arXiv](https://arxiv.org/abs/2103.02142), [ResearchGate](https://www.researchgate.net/publication/349758949_Learning_to_Fly-a_Gym_Environment_with_PyBullet_Physics_for_Reinforcement_Learning_of_Multi-agent_Quadcopter_Control))

CoolCoder54323, 2021-05-31:
This function works perfectly and the quick start guide is the documentation I was looking for, but I think you should put a link to it in the readme it would have saved me at least a lot of time.

Thanks for the quick response!

closed 2021-06-18 by CoolCoder54323