learnsyslab/gym-pybullet-drones #52: [Question] Frequency

CoolCoder54323, 2021-06-19:
What does the freq parameter do because when I set it from 240 to 1 it makes no difference in the amount of time it takes to complete a step. Also in what scenario would you want to change aggregate_phy_steps from 1?

JacopoPan, 2021-06-19:
Hi @CoolCoder54323,

argument `freq` sets `self.SIM_FREQ` which is the inverse of the timestep PyBullet advances the simulation at each call to `p.stepSimulation()`.

`aggregate_phy_steps` simply determines how many `p.stepSimulation()` happen within each call to `env.step()` (it saves a number of read/writes and some wall-clock, it is also useful if you want more accuracy in the simulation, with a higher PyBullet frequency but a fixed or coarser control step frequency).

CoolCoder54323, 2021-06-20:
So shouldn't the simulation slow down if you set freq to 1

JacopoPan, 2021-06-20:
No, setting `freq=1` simply means that every call to `p.stepSimulation()` (one per `env.step()` iff `aggregate_phy_steps==1`) will advance the simulation, in the simulation's clock, by 1 second. This can happen faster that 1 second in wall clock (the one of the world you live in).

The idea is to run simulations faster than real-time, to collect more data, you can check PyBullet documentation for using it as a gaming engine, with (best effort) real-time stepping, but this is not the goal in this repo.

In general, `freq=1` is not a good idea because it would result in poor physics accuracy for fast dynamics/collisions.

CoolCoder54323, 2021-06-20:
What I was trying to do is make each action of the AI last longer in the simulation. So if I'm understanding this correctly I would do that using `aggregate_phy_steps` so it updates `p.stepSimulation()` more than once using the same action.

jacopo-panerati, 2021-06-20:
What I’d say you are doing is to make the control frequency coarser than the physics engine stepping frequency, and yes, that is done with aggregate_physics_steps

E.g. if PyBullet freq is 240Hz, with aggregate_physics_steps set to 5 means controlling the quadrotor at 48Hz

closed 2021-06-21 by CoolCoder54323