learnsyslab/gym-pybullet-drones #54: Episode vs timesteps

nbenave, 2021-07-19:
Hi there,
Thank you again for you amazing contribution. I learned a lot from this project.

1) how this project defines an EPISODE ?
2) is EPISODE equivalent to one flight ?
3) is EPISODE defines as few flights until a reward coverage ?

4) in singleagent.py the model.learn method receive total_timesteps as parameter.
also, there's a stopTrainingRewardThreshold callback with constant reward threshold of -0

5) can you clarify how this model learn in units of steps/episode/timesteps ?
6) there's any way to review the learning process in episodes iterations and not time timesteps ?

Thanks,

JacopoPan, 2021-07-19:
Hi again @nbenave,

"episodes" are 5 seconds (in simulation clock) flights, in most aviaries in this repo. Of course that is arbitrary and application dependent. Once the episode is over, `env.step()` will return `done=True`.

The `stopTrainingRewardThreshold` set at 0 is an upperbound that, in the current implementation cannot be achieved (but again, that is application/implementation/reward shaping dependent as well).

W.r.t. 5 and 6, that depends on how trajectories/batches are used by each RL algorithm. You should look into the implementations of SAC, PPO, etc. in `stable-baselines3`.

sunjeet-jena, 2021-11-02:
Is there a way we can increase (in simulation clock) time ?

JacopoPan, 2021-11-02:
Do you mean the episodes length in seconds, @sunjeet-jena?
There is attribute `self.EPISODE_LEN_SEC` in both `BaseSingeleAgentAviary` and `BaseMultiagentAviary` classes.

closed 2022-08-03 by JacopoPan