learnsyslab/gym-pybullet-drones #60: fly.py is slow with vision = True (VisionAviary)

thaipduong, 2021-11-19:
Hi,
I'd like to get depth image from the drones and have a few questions.
I try fly.py with VisionAviary object by setting vision = True. However, the GUI shows that the drones move very slowly, even thought the camera image size is 64x48 only. Is this expected? Is there any way to speed this up? And is there a laser scanner available so that I can get the depth measurements faster?

Thank you so much.

JacopoPan, 2021-11-19:
Hi @thaipduong 

yes, that is expected, of course, the slowdown will be proportional to:
- number of UAVs `num_drones` (Not the the vision is in each UAV frame so these are all separate render views)
- resolution `self.IMG_RES = np.array([64, 48])` in `BaseAviary`
- framerate of capture `self.IMG_FRAME_PER_SEC = 24` in `BaseAviary`
For the sake of collecting data, I would recommend running headless (`gui=False`) and to use an Ubuntu machine and GPU
In this case you can uncomment line 203 in `BaseAviary`
```
if platform == "linux": p.setAdditionalSearchPath(pybullet_data.getDataPath()); plugin = p.loadPlugin(egl.get_filename(), "_eglRendererPlugin"); print("plugin=", plugin)
```
for hardware acceleration (see section "Plugins" of [PyBullet manual](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3))

thaipduong, 2021-11-19:
I see. Thank you, @JacopoPan 
Just to confirm, there is currently no lidar on the drone, right?

JacopoPan, 2021-11-19:
Sorry, I forgot to answer your second question.

I didn't implement anything specifically in this repo, but if you look at another work-in-progress env I have [here](https://github.com/JacopoPan/gym-marl-reconnaissance/blob/6a9faa314106ab1ccd07cf3ff2076daa9e996579/gym_marl_reconnaissance/envs/recon_arena.py#L709) there is an example of how to use [PyBullet's `rayTest`](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.e7a8kr2734k2) to determine if there is line-of-sight between agents.

thaipduong, 2021-11-19:
Awesome! I'll take a look. Thanks, @JacopoPan

closed 2021-11-19 by thaipduong