learnsyslab/gym-pybullet-drones #61: Subprocess issue

Trad-taha-yacine, 2021-11-20:
Hi there,
Thank's for your amazing contribution and putting together such an awesome project,

I have been testing all of your codes and algorithmes which all works fine for me exept singleagent.py and multi-agent.py where i had this error, hopefully u can help me with : 

Traceback (most recent call last):
  File "C:\Users\yacine\Desktop\gym-pybullet-drones-master\gym-pybullet-drones-master\experiments\learning\multiagent.py", line 157, in <module>
    git_commit = subprocess.check_output(["git", "describe", "--tags"]).strip()
  File "C:\Users\yacine\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 419, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\yacine\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 500, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\yacine\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 946, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\yacine\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1413, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Le fichier spécifié est introuvable

Ps : I tried to comment : 

  git_commit = subprocess.check_output(["git", "describe", "--tags"]).strip()
    with open(filename+'/git_commit.txt', 'w+') as f:
        f.write(str(git_commit))

and it works for the single agent but didnt for multi_agent, 

Thank's

JacopoPan, 2021-11-21:
Hi @Trad-taha-yacine,

you can most certainly comment out those lines as they are nothing more than a print in a log file of the commit of the code you are executing (for diagnostic/debugging purposes).

Do you get another error thrown by the `multiagent.py` script once you have removed those?

Trad-taha-yacine, 2021-11-24:
Hi, Thank's for your respond,
I tried to do the same thing with multiagent code, it dosn't throw any error, the code is being executed but with no respond or anything, even with small timesteps_total = 1000.
the process doesn't terminate, i dont know what the code is exactly doing,
Thank you for your help.

JacopoPan, 2021-11-24:
I see. Can you try `pip3 install --upgrade 'ray[rllib]'` and copy your terminal output (of `multiagent.py`, `test_multiagent.py`) here? The scripts still run for me but certainly there have been several changes and deprecations in RLlib since my last commit.

Trad-taha-yacine, 2021-11-26:
Hi, 
I think the solution you gave me worked, after i upgraded 'ray[rllib]', the code worked fine, thank you so much.
Nice work again.

closed 2021-11-26 by JacopoPan