learnsyslab/gym-pybullet-drones #63: Simulation in the presence of a wind field.

dipaco, 2021-11-26:
Given a wind field f: R^3xt -> R^3, is it possible to simulate how the robots would behave in it? Thanks.

JacopoPan, 2021-11-26:
Hi @dipaco,

can you help me understand what you want to implement? Could a time-varying force contribution do the trick? The way the robot model is specified in the URDF file/simulation, one could apply external forces on its center of mass or at the site of each motor.

dipaco, 2021-11-26:
Thank you @JacopoPan for your quick answer. This might work, I need to look into it, but so far it looks pretty much like what I need. Yes, what I meant by f: R^3 x t -> R^3 is a time-varying force that gives me a wind velocity at every position x. I can convert that wind field into a drag force using some knowledge of the geometry of the quadcopter. However, I wasn't sure if it was possible to exert that external force into the quadcopter at simulation time. I will try your example. 

Thank you so much.

JacopoPan, 2021-11-26:
No prob.
Note that you should call `p.applyExternalForce()` before every `p.stepSimulation()`.
`env.step()` might contain multiple `p.stepSimulation()`, if you created `env` with argument `aggregate_phy_steps` > 1.
In that case, you might want to modify `BaseAviary` directly, to apply the external force just before `p.stepSimulation()`.

chijiokekechi, 2022-03-11:
Hi @JacopoPan, thank you for answering this. Just for clarification, the `p.applyExternalForce()` adds this force to the total forces acting on the drone. What units are the forces and how do I distinguish this from the drones internal forces (i.e. thrust and the likes)? Thank you very much.

JacopoPan, 2022-03-11:
Hi @chijiokekechi,

I would refer you to the [PyBullet manual](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.mq73m9o2gcpy) for the use of `applyExternalForce`. The unit is Newton. In the method call, you would specify one of the links composing the robots (4 for the center of the drone).