learnsyslab/gym-pybullet-drones #67: Support for external radio controllers [Enhancement]

Sridhar701Gatech, 2022-02-13:
Considering that drones are controlled using a radio controller in real life, it would be a great addition to add support for an external radio controller to control drones in the pybullet simulator. Connection shouldn't be an issue since these controllers come with a wired connection to a PC. (Similar to ones used in FPV racing simulators. Example: [https://www.velocidrone.com/](https://www.velocidrone.com/))

- It would also be great to add binds for additional switches on the radio controller to enable/disable a custom control script such as when one wants to trigger an RL control algorithm to take over the manual control while in flight.

JacopoPan, 2022-02-14:
Hi @Sridhar701Gatech 

That's a good point, it wasn't originally within my plans but, later in the year, you will see a bridge for `gym-pybullet-drones` or [`safe-control-gym`](https://github.com/utiasDSL/safe-control-gym) and a [CrazySwarm](https://github.com/USC-ACTLab/crazyswarm)-like system for a virtual + sim2real IROS 2022 competition.

imamim, 2022-03-02:
Wow, that will be some seriously nice work :smile:  I wish you success in advance.
I would like to ask, when do you foresee the publication of the specified system? The deadline for IROS2022 seems to be 31st July but I wanted to ask anyway.

JacopoPan, 2022-03-02:
Thanks @imamim, I also have plenty of good hopes but I know it will include a few challenging steps!
The sim2real component is only planned for the time of the competition so the goal would be to have it ready by end of the summer.

imamim, 2022-03-03:
Understood. Yes, it can be a difficult process.Thank you for the information, good luck :)

closed 2022-08-03 by JacopoPan