learnsyslab/gym-pybullet-drones #68: rpy_rates in Explicit dynamics implementation

gokhanalcan, 2022-03-03:
Hi,

I have a question regarding your Explicit dynamics implementation.

Is the variable `rpy_rates` here https://github.com/utiasDSL/gym-pybullet-drones/blob/a0534f8a28e9b47f1d9912223952075a032be455/gym_pybullet_drones/envs/BaseAviary.py#L832 for Body-axis rates or Euler-angle rates? 

In the [paper](https://arxiv.org/pdf/2103.02142.pdf) just before Eq. 2, `rpy_rates_deriv` is written as "_change in the turn rates in the local frame_". So I thought it refers to body-axis. However, after integrating and obtaining `rpy` as: https://github.com/utiasDSL/gym-pybullet-drones/blob/a0534f8a28e9b47f1d9912223952075a032be455/gym_pybullet_drones/envs/BaseAviary.py#L855 it is used as Euler angles here: https://github.com/utiasDSL/gym-pybullet-drones/blob/a0534f8a28e9b47f1d9912223952075a032be455/gym_pybullet_drones/envs/BaseAviary.py#L859 without [converting](http://stengel.mycpanel.princeton.edu/Quaternions.pdf) it from local to global. 

Am I missing something? Could you please help me clarify?

Thanks in advance

JacopoPan, 2022-03-03:
Hi @gokhanalcan,

going by memory, I'd say your first assumption is the right one and line 859 might be wrong. However, the main reason for the call to `resetBasePositionAndOrientation` in line 857 is just rendering and data storage in between simulation steps. I'm now wondering if it can have cascade effects: did you develop an example in which you obtained diverging results between the Bullet simulation and the explicit dynamics implementation (which, admittedly, I haven't used in a while)?

gokhanalcan, 2022-03-04:
Hi @JacopoPan ,

Thank you for your detailed responses. I don't think that it is just related to rendering.
Because `_updateAndStoreKinematicInformation` function get `quat` from Bullet here:
https://github.com/utiasDSL/gym-pybullet-drones/blob/a0534f8a28e9b47f1d9912223952075a032be455/gym_pybullet_drones/envs/BaseAviary.py#L512
Then, using `quat`, rotation matrix is calculated as:
https://github.com/utiasDSL/gym-pybullet-drones/blob/a0534f8a28e9b47f1d9912223952075a032be455/gym_pybullet_drones/envs/BaseAviary.py#L833
and `rotation` is used to transform thrust to world frame.
https://github.com/utiasDSL/gym-pybullet-drones/blob/a0534f8a28e9b47f1d9912223952075a032be455/gym_pybullet_drones/envs/BaseAviary.py#L837
So it affects the dynamics as well, doesn't it?

JacopoPan, 2022-03-04:
That's possible but I would need to check that numerically because static analysis could be misleading (it would visibly come up when the x-axis of the quadrotor's frame is not aligned to the x-axis world frame, I think). That's why I asked if you already had an example that led to incongruous results. Otherwise, I'll have to devise one and I could do it only in the next few weeks.

JohnstonXi, 2022-03-24:
I was also confused with frame conversion as I was reading SimplePIDControl.py.

This line converts `target_force` (world frame) to `thrust` (local frame)
https://github.com/utiasDSL/gym-pybullet-drones/blob/a4e165bcbeb9133bee4bf920fca3d1a170f7bba7/gym_pybullet_drones/control/SimplePIDControl.py#L185

It seems a mistake because this rotation matrix should convert from local to world frame, not the other way round.

JacopoPan, 2022-03-24:
Hi @JohnstonXi,

`SimplePIDControl` is a "legacy" implementation (based on https://github.com/prfraanje/quadcopter_sim) but it is not used (nor recommended for) with the Crazyflie model.

As I mentioned for the issue raised by @gokhanalcan, if you have the time to create a complete bug report (what does not work as expected), fix (what the code should do instead), and counter example (a MWE that didn't work before and works after the fix), I'd love to incorporate it to the codebase. Otherwise I will attend it as soon as I have some more time at hand.

closed 2022-08-03 by JacopoPan