learnsyslab/gym-pybullet-drones #69: error in _normalizedActionToRPM in BaseAviary.py (line 891)

Sridhar701Pitt, 2022-03-04:
```
# The following line has incorrect mapping for action > 0 (See BaseAviary.py (line 891))

np.where(action <= 0, (action+1)*self.HOVER_RPM, action*self.MAX_RPM) # Non-linear mapping: -1 -> 0, 0 -> HOVER_RPM, 1 -> MAX_RPM

# The correct one should be:

np.where(action <= 0, (action+1)*self.HOVER_RPM, self.HOVER_RPM + (self.MAX_RPM - self.HOVER_RPM)*action) # Non-linear mapping: -1 -> 0, 0 -> HOVER_RPM, 1 -> MAX_RPM`
```

JacopoPan, 2022-03-05:
Thanks for noting this, there was also a misplaced parenthesis at line 889—`np.any(np.abs(action)) > 1` instead of `np.any(np.abs(action) > 1)`

As a general suggestion 😄, it would make it a lot easier to review such bug reports if you could attach a small test script demonstrating the difference between implementations like this one

```
import numpy as np
import matplotlib.pyplot as plt

def test(
         action,
         option
         ):
        # step_counter = -1
        HOVER_RPM = 1000
        MAX_RPM = 5000
        if np.any(np.abs(action) > 1):
            # print("\n[ERROR] it", step_counter, "in BaseAviary._normalizedActionToRPM(), out-of-bound action")
            return np.array([-1,-1,-1,-1])
        if option == 'a':
            return np.where(action <= 0, (action+1)*HOVER_RPM, action*MAX_RPM) # Non-linear mapping: -1 -> 0, 0 -> HOVER_RPM, 1 -> MAX_RPM
        elif option == 'b':
            return np.where(action <= 0, (action+1)*HOVER_RPM, HOVER_RPM + (MAX_RPM - HOVER_RPM)*action) # Non-linear mapping: -1 -> 0, 0 -> HOVER_RPM, 1 -> MAX_RPM`

def main():
    fig, (ax1, ax2) = plt.subplots(1, 2)
    x = np.linspace(-2, 2, 100)
    plot1 = np.zeros(100)
    for i in range(len(plot1)):
        ret = test(np.array([x[i], x[i], x[i], x[i]]), 'a')
        plot1[i] = ret[0]
    ax1.plot(x, plot1)
    plot2 = np.zeros(100)
    for i in range(len(plot2)):
        ret = test(np.array([x[i], x[i], x[i], x[i]]), 'b')
        plot2[i] = ret[0]
    ax2.plot(x, plot2)
    plt.show()

if __name__ == "__main__":
    main()
```
[image]

JacopoPan, 2022-03-05:
See commit [a4e165b](https://github.com/utiasDSL/gym-pybullet-drones/commit/a4e165bcbeb9133bee4bf920fca3d1a170f7bba7)

closed 2022-03-05 by JacopoPan