learnsyslab/gym-pybullet-drones #70: About Instantaneously Control of Drones

imamim, 2022-03-10:
Hi, have a good day

I have a few confusions about instantaneously controlling drones and I want to ask about it. If I need to talk about what I want to reach, it is to be able to control the movements of the drone using the angular and linear velocity values ​​that I set myself. That values are the input, then the needed output is movement.

First of all, when I looked at the past issues, I saw that the same topic was discussed on #49. You mentioned that PyBullet's `p.resetBasePositionAndOrientation()` and `p.resetBaseVelocity()` functions can be used here.When I use these functions, do we include the explicit dynamic model that you implement, or do we only use PyBullet dynamics?

Another thing that comes to my mind is; When I examine the examples in the `assignments/` folder, a control is processed using `obs` values obtained from `env.step()` and then a simulation step is performed with `env.step(ACTION)`. Here ACTION corresponds to engine RPM values. Although the system structure makes sense, it doesn't make much sense for me to go through the RPM account.

Should I use `BaseAviary.py`, `ControlAviary.py` or `control/Control.py` instead? How can I control drones as I mentioned above ?

Thanks your answer from now :)

JacopoPan, 2022-03-11:
Hello @imamim,

is you question how to instantaneously change/set linear and/or angular velocity to arbitrary values?
I would say that PyBullet's `p.resetBaseVelocity()` method is what you are looking for but, or course, that defies a good part of the purpose of this simulation because you wouldn't adhere anymore to the dynamics of the quadrotor (imposed by its mass, thrust, inertia, torque coefficients, etc.)

imamim, 2022-03-12:
Yeah, I understood this issue while working the code yesterday,`p.resetBaseVelocity() `is working properly. But here was the problem. It was necessary to run the simulation` env.step(action)`.In the step phase, I always give the action value a value that I set as a constant. Here I observed that this action has no effect after `p.resetBaseVelocity`, with almost sure (If i'm not escape anything).  Also, after giving a random value (and do the step), I get the `obs` value with the `CtrlAviary.py/_computeObs()` function.

I am almost certain that these values ​​are correct. The strangeness here was confusing me, but the problem is gone for now, thanks for your answer :)

closed 2022-03-12 by imamim