learnsyslab/gym-pybullet-drones #72: Error when runing singleagent.py last part: print training progression

MullerLin, 2022-03-20:
Hi, 
I am trying to modify some tasks for single agent, 
but I met a error when running the last part of singleagent.py: ##print training progression##:

    #### Print training progression ############################
    with np.load(filename+'/evaluations.npz') as data:
        for j in range(data['timesteps'].shape[0]):
            print(str(data['timesteps'][j])+","+str(data['results'][j][0][0]))

The error reported is as follow:

Traceback (most recent call last):
  File "/home/experiments/learning/singleagent.py", line 261, in <module>
    print(str(data['timesteps'][j])+","+str(data['results'][j][0][0]))
IndexError: invalid index to scalar variable.

Through the code can still work well and train the agent, I am wandering how to fix the error, but I have no idea about  '/evaluations.npz' from this python file. Could you help me to  deal with such error and thank you so much.

Thanks a lot for your great projects.

JacopoPan, 2022-03-21:
Hi @MullerLin 

`evaluations` is created by the RL agent and `.npz` is simply NumPy compressed file export format.
You are receiving an error because its content is not in the exact form that the script expects (which is totally plausible, if you changed the training scripts).

Just print out the content of the file with something like:
```
from numpy import load

data = load('evaluations.npz')
lst = data.files
for item in lst:
    print(item)
    print(data[item])
```

MullerLin, 2022-03-24:
> 
 It works now. Thanks ya lot for your help!

closed 2022-03-24 by JacopoPan