learnsyslab/gym-pybullet-drones #74: MultiAgent RGB Observation

gauravkuppa, 2022-03-23:
When I try to run `python3 multiagent.py --num_drones 4 --obs rgb`, I get the error `[ERROR] ObservationType.RGB for multi-agent systems not yet implemented`. However, in the results from the README, I can see that you tested multiple drones with vision. Is this feature available?

  | Lenovo P52 (i7-8850H/Quadro P2000) | 2020 MacBook Pro (i7-1068NG7)
-- | -- | --
Rendering | OpenGL | CPU-based TinyRenderer
Single drone, no vision | 15.5x | 16.8x
Single drone with vision | 10.8x | 1.3x
Multi-drone (10), no vision | 2.1x | 2.3x
Multi-drone (5) with vision | 2.5x | 0.2x
80 drones in 4 env, no vision | 0.8x | 0.95x

If not, I am willing to contribute to this repo to add this functionality. What is the status of this work?

JacopoPan, 2022-03-23:
Hi @gauravkuppa 

you can already use `ObservationType.RGB` with `BaseMultiagentAviary` and all subclasses.
What is **not** implemented yet is a learning script (like `multiagent.py`) with an RL agent intended to handle that type of observation. Does this answer your question?

gauravkuppa, 2022-03-30:
Could you point me towards how to implement multiagent learning?

JacopoPan, 2022-03-30:
Do you have a MARL approach in mind? The `multiagent.py` is meant as an example leveraging the central critic/decentralized polices example in RLlib but that's a very specific and arbitrary choice.