learnsyslab/gym-pybullet-drones #76: Some questions about the logger in test_multiagent/test_singleagent.py

MullerLin, 2022-03-28:
Hi, thanks to your wonderful project, 
I am working on training some RL tasks on the project, and I use test_multiagent.py/test_singleagent.py to see the training results, like how the drone act. 
I find that the final plotted data curves from logger are clipped and normalized because the logger logs the data from _obs_ , the data after  clipping and normalizing. It made me confused at first when I tried to trained the drone to fly some certain position, and I modified data on the curve by multiplying the normalization coefficient. 

I am wondering whether there are some reasons to show the clipped and normalized data on final curve in the two test.py, like helping analyze or some other reasons.

Thanks a lot.

JacopoPan, 2022-03-29:
Hi @MullerLin,

glad you enjoy the project! The clipping/normalization of abs and actions obviously is intended to help the stock RL agents in stable-baselines3 to more easily learn. There isn't any specific reason for the plots to follow the same clipping/normalization other than the simplicity of using the data as they come out of the aviary environment class.

MullerLin, 2022-04-17:
> Hi @MullerLin,
>
> glad you enjoy the project! The clipping/normalization of abs and actions obviously is intended to help the stock RL agents in stable-baselines3 to more easily learn. There isn't any [...]

Thanks a lot for your reply! It helps me to understand the project better!