learnsyslab/gym-pybullet-drones #77: Linearized state space model

brunzema, 2022-04-13:
I have a very short question regarding the `BaseAviary`. I want to implement a simple optimal controller (LQR) as a baseline and for this, I need the state space model. Are the attributes `self.A` and `self.B` obtained by setting `dynamics_attributes =True` the matrices of the linearized state space model (code snippet below)? If yes, the states are probably position and velocity in x, y direction correct?

```python

class BaseAviary(gym.Env):
     def __init__(self, ..., dynamics_attributes)
        ...

        #### Create attributes for dynamics control inputs #########
        self.DYNAMICS_ATTR = dynamics_attributes
        if self.DYNAMICS_ATTR:
            if self.DRONE_MODEL == DroneModel.CF2X:
                self.A = np.array([ [1, 1, 1, 1], [1/np.sqrt(2), 1/np.sqrt(2), -1/np.sqrt(2), -1/np.sqrt(2)], [-1/np.sqrt(2), 1/np.sqrt(2), 1/np.sqrt(2), -1/np.sqrt(2)], [-1, 1, -1, 1] ])
            elif self.DRONE_MODEL in [DroneModel.CF2P, DroneModel.HB]:
                self.A = np.array([ [1, 1, 1, 1], [0, 1, 0, -1], [-1, 0, 1, 0], [-1, 1, -1, 1] ])
            self.INV_A = np.linalg.inv(self.A)
            self.B_COEFF = np.array([1/self.KF, 1/(self.KF*self.L), 1/(self.KF*self.L), 1/self.KM])
```

Thank you very much!

JacopoPan, 2022-04-14:
Hello @brunzema,

no, that's not really the intention, those matrices/vector that can be used to compute RPMs as non-negative least squares from desired torques and thrust. Could you explain a bit further what you are doing/require? For LQR in a 2D (x-z plane) quadcopter you can look at our other repo [`safe-control-gym`](https://github.com/utiasDSL/safe-control-gym), (e.g. these [implementations](https://github.com/utiasDSL/safe-control-gym/tree/main/safe_control_gym/controllers/lqr)).

brunzema, 2022-04-19:
Hey @JacopoPan,

thank you for the fast reply! Yeah, I looked at the matrices and more and more realized, that it was not like I thought. Makes sense. :) Indeed, a 2D example is sufficient for my use-case. I want to tune controller parameters with Bayesian optimization but with a time-varying control objective.  I will continue to use your other repo. Thank you for all the implementations, they are super useful!

closed 2022-04-19 by brunzema

iitimii, 2024-12-17:
Is there a way to get the linearized state space model with **quaternions**? I'm also trying to design an LQR controller for the cf2x. [https://github.com/utiasDSL/safe-control-gym/blob/main/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py](safe-control-gym) uses Euler angles.

JacopoPan, 2024-12-17:
@iitimii could you please open an issue in https://github.com/utiasDSL/safe-control-gym/issues for that?

iitimii, 2024-12-17:
Alright, thank you