learnsyslab/gym-pybullet-drones #78: Error in installing gym-pybullet-drones.

HimGautam, 2022-04-21:
I tried installing the package as given in the [readme.md](https://github.com/utiasDSL/gym-pybullet-drones#readme). It started normally and updated or downloaded the necessary packages required but after some time it gave me the error. Can you please tell me what is wrong with it and how can I rectify it. My OS is Ubuntu 20.04 lts, my machine has a gpu GTX 1650.

```
Collecting torch>=1.8.1
  Downloading torch-1.11.0-cp39-cp39-manylinux1_x86_64.whl (750.6 MB)
     |████████████████████████████████| 750.6 MB 22 kB/s 
Requirement already satisfied: pandas in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from stable-baselines3<2.0,>=1.5->gym-pybullet-drones==1.0.0) (1.4.1)
Requirement already satisfied: typing-extensions in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from torch>=1.8.1->stable-baselines3<2.0,>=1.5->gym-pybullet-drones==1.0.0) (4.2.0)
Requirement already satisfied: setuptools in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from jsonschema->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (61.2.0)
Requirement already satisfied: pyrsistent>=0.14.0 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from jsonschema->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (0.18.0)
Requirement already satisfied: pytz>=2020.1 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from pandas->stable-baselines3<2.0,>=1.5->gym-pybullet-drones==1.0.0) (2021.3)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from requests->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (1.26.8)
Requirement already satisfied: charset-normalizer~=2.0.0 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from requests->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (2.0.4)
Requirement already satisfied: idna<4,>=2.5 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from requests->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (3.3)
Requirement already satisfied: certifi>=2017.4.17 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from requests->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (2021.10.8)
Requirement already satisfied: platformdirs<3,>=2 in /home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages (from virtualenv->ray<2.0,>=1.8->gym-pybullet-drones==1.0.0) (2.5.2)
Collecting distlib<1,>=0.3.1
  Downloading distlib-0.3.4-py2.py3-none-any.whl (461 kB)
     |████████████████████████████████| 461 kB 2.8 MB/s 
Building wheels for collected packages: gym
  Building wheel for gym (setup.py) ... done
  Created wheel for gym: filename=gym-0.21.0-py3-none-any.whl size=1616826 sha256=92b8584a861541c0263d2cb2cbb0db61acb359982db442db183a108b527ba1be
  Stored in directory: /home/himanshu/.cache/pip/wheels/b3/50/6c/0a82c1358b4da2dbd9c1bb17e0f89467db32812ab236dbf6d5
Successfully built gym
Installing collected packages: numpy, filelock, distlib, cycler, virtualenv, torch, msgpack, gym, stable-baselines3, scipy, ray, gym-pybullet-drones
  Attempting uninstall: numpy
    Found existing installation: numpy 1.19.5
    Uninstalling numpy-1.19.5:
      Successfully uninstalled numpy-1.19.5
  Attempting uninstall: cycler
    Found existing installation: cycler 0.11.0
    Uninstalling cycler-0.11.0:
      Successfully uninstalled cycler-0.11.0
  Attempting uninstall: gym
    Found existing installation: gym 0.23.0
    Uninstalling gym-0.23.0:
      Successfully uninstalled gym-0.23.0
  Attempting uninstall: scipy
    Found existing installation: scipy 1.7.3
    Uninstalling scipy-1.7.3:
      Successfully uninstalled scipy-1.7.3
  Running setup.py develop for gym-pybullet-drones
    ERROR: Command errored out with exit status 1:
     command: /home/himanshu/anaconda3/envs/myenv/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/himanshu/gym-pybullet-drones/setup.py'"'"'; __file__='"'"'/home/himanshu/gym-pybullet-drones/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: /home/himanshu/gym-pybullet-drones/
    Complete output (3 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/himanshu/anaconda3/envs/myenv/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/himanshu/gym-pybullet-drones/setup.py'"'"'; __file__='"'"'/home/himanshu/gym-pybullet-drones/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.
```

HimGautam, 2022-04-21:
The issue got solved, turns out I had outdated version of setuptools. Once I updated the setuptools module, the pybullet-drones package installed smoothly.

closed 2022-04-21 by HimGautam

JacopoPan, 2022-04-21:
Sounds good, I was about to suggest to try
```
sudo apt-get install python3-setuptools
```