learnsyslab/gym-pybullet-drones #79: Simulator crashes when I run fly.py

HimGautam, 2022-04-21:
After installing fly.py, I tested the simulator by fly.py file. Simulation ran for some time and then crashed automatically. The following message appeared on the Terminal.

```
(myenv) himanshu@himanshu-Lenovo-Legion-Y540-15IRH-PG0:~/gym-pybullet-drones/examples$ python3 fly.py  
pybullet build time: Mar 26 2022 03:01:35
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA GeForce GTX 1650/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 510.54
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 510.54
Vendor = NVIDIA Corporation
Renderer = NVIDIA GeForce GTX 1650/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999701976776, 0.4330126941204071, 0.0, 0.0, 0.8660253286361694, 0.4999999701976776, 0.0, -0.4999999701976776, 0.4330126643180847, -0.75, 0.0, -0.0, 5.960464477539063e-08, -2.999999761581421, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
ven = NVIDIA Corporation
ven = NVIDIA Corporation

[INFO] BaseAviary.render() ——— it 0005 ——— wall-clock time 1.6s, simulation time 0.0s@240Hz (0.01x)
[INFO] BaseAviary.render() ——— drone 0 ——— x +00.00, y +00.00, z +00.10 ——— velocity +00.00, +00.00, -00.20 ——— roll +00.00, pitch -00.00, yaw +00.00 ——— angular velocity +0.0000, +0.0000, +0.0000 ——— 
[INFO] BaseAviary.render() ——— drone 1 ——— x -00.26, y -00.15, z +00.15 ——— velocity +00.00, +00.00, -00.20 ——— roll -00.00, pitch -00.00, yaw +30.00 ——— angular velocity +0.0000, +0.0000, +0.0000 ——— 
[INFO] BaseAviary.render() ——— drone 2 ——— x -00.26, y -00.45, z +00.20 ——— velocity +00.00, +00.00, -00.20 ——— roll -00.00, pitch -00.00, yaw +60.00 ——— angular velocity +0.0000, +0.0000, +0.0000 ——— 

[INFO] BaseAviary.render() ——— it 0245 ——— wall-clock time 2.6s, simulation time 1.0s@240Hz (0.40x)
[INFO] BaseAviary.render() ——— drone 0 ——— x -00.10, y -00.02, z +00.10 ——— velocity -00.14, -00.08, +00.00 ——— roll +01.29, pitch +00.61, yaw -00.69 ——— angular velocity +0.4407, +0.0317, -0.1197 ——— 
[INFO] BaseAviary.render() ——— drone 1 ——— x -00.29, y -00.24, z +00.15 ——— velocity -00.01, -00.19, +00.00 ——— roll +00.79, pitch +00.86, yaw +30.05 ——— angular velocity -0.9583, +0.2403, +0.0848 ——— 
[INFO] BaseAviary.render() ——— drone 2 ——— x -00.19, y -00.52, z +00.20 ——— velocity +00.15, -00.11, +00.00 ——— roll +00.35, pitch -00.12, yaw +60.63 ——— angular velocity -0.2727, -0.1292, -0.0395 ——— 

[INFO] BaseAviary.render() ——— it 0485 ——— wall-clock time 3.6s, simulation time 2.0s@240Hz (0.56x)
[INFO] BaseAviary.render() ——— drone 0 ——— x -00.26, y -00.14, z +00.10 ——— velocity -00.11, -00.16, +00.00 ——— roll +01.35, pitch +01.02, yaw -00.46 ——— angular velocity +0.1272, -0.1216, +0.0596 ——— 
[INFO] BaseAviary.render() ——— drone 1 ——— x -00.27, y -00.43, z +00.15 ——— velocity +00.08, -00.17, -00.00 ——— roll +01.30, pitch +00.58, yaw +30.25 ——— angular velocity -1.0060, +0.1241, +0.1332 ——— 
[INFO] BaseAviary.render() ——— drone 2 ——— x -00.02, y -00.59, z +00.20 ——— velocity +00.20, -00.01, +00.00 ——— roll -00.53, pitch +00.63, yaw +60.92 ——— angular velocity -0.3421, -0.5151, +0.0179 ——— 

[INFO] BaseAviary.render() ——— it 0725 ——— wall-clock time 4.6s, simulation time 3.0s@240Hz (0.66x)
[INFO] BaseAviary.render() ——— drone 0 ——— x -00.31, y -00.31, z +00.10 ——— velocity -00.00, -00.16, -00.00 ——— roll +00.85, pitch +00.71, yaw +00.09 ——— angular velocity +0.2404, +0.7939, -0.0123 ——— 
[INFO] BaseAviary.render() ——— drone 1 ——— x -00.14, y -00.57, z +00.15 ——— velocity +00.15, -00.12, -00.00 ——— roll -00.42, pitch -00.02, yaw +30.68 ——— angular velocity -1.1798, +0.2701, -0.0612 ——— 
[INFO] BaseAviary.render() ——— drone 2 ——— x +00.17, y -00.56, z +00.20 ——— velocity +00.18, +00.09, +00.00 ——— roll -01.46, pitch -00.88, yaw +60.04 ——— angular velocity -0.4177, -0.6529, +0.0062 ——— 

[INFO] BaseAviary.render() ——— it 0965 ——— wall-clock time 5.6s, simulation time 4.0s@240Hz (0.72x)
[INFO] BaseAviary.render() ——— drone 0 ——— x -00.26, y -00.49, z +00.10 ——— velocity +00.11, -00.15, -00.00 ——— roll -00.78, pitch +00.21, yaw -00.01 ——— angular velocity -0.5759, -0.4916, -0.0195 ——— 
[INFO] BaseAviary.render() ——— drone 1 ——— x +00.04, y -00.61, z +00.15 ——— velocity +00.19, +00.00, +00.00 ——— roll -00.08, pitch +01.50, yaw +30.26 ——— angular velocity -0.7561, -0.2495, -0.0186 ——— 
[INFO] BaseAviary.render() ——— drone 2 ——— x +00.29, y -00.42, z +00.20 ——— velocity +00.08, +00.18, -00.00 ——— roll -01.41, pitch -00.42, yaw +60.52 ——— angular velocity +0.0439, +0.0438, -0.0650 ——— 
[... 37 lines cut ...]
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
QObject::moveToThread: Current thread (0x56470a0563f0) is not the object's thread (0x564709fa2570).
Cannot move to target thread (0x56470a0563f0)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/himanshu/anaconda3/envs/myenv/lib/python3.9/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl.

Aborted (core dumped)
```

JacopoPan, 2022-04-22:
Hi @HimGautam 

if you look at the error message, it does clarify what led to the crash (`Could not load the Qt platform plugin "xcb"`) and possible solutions. From the rest of the output, it seems to me that this happens **after** the PyBullet GUI is closed and the scripts is creating summary plots with `matplotlib`.

The first few things I'd try would be reinstalling the non-loadable plugin, matplotlib, or eventually PyBullet.

HimGautam, 2022-04-22:
I've used ```pip uninstall PyQt5 ``` and ```conda install pyqt``` to reinstall Qt. The error message seems to have disappeared. But still simulation stops automatically and a window with graphs pops up. So, does fly.py runs the simulation for 2-3 seconds?

JacopoPan, 2022-04-22:
It should run for about 12’’ and that should be reflected in the x axes of the plots. Is it not?

HimGautam, 2022-04-22:
Yes, fly.py ran for 12 units and the velocity.py ran for 5 units.

JacopoPan, 2022-04-22:
That’s seems about right. If you inspect the scripts, it should be easy to see how to change the time duration (`--duration`)

HimGautam, 2022-04-23:
Ok thanks

HimGautam, 2022-04-23:
Is there a tutorial guide or a python file where I can learn about what is happening under the hood at the level of pybullet?

JacopoPan, 2022-04-23:
All calls TO PyBullet's client are in [`BaseAviary.py`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/envs/BaseAviary.py)
To better understand them, refer to [PyBullet user guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#)
To peek inside Bullet, you can look at [its code and recompile from source](https://github.com/bulletphysics/bullet3)

closed 2022-04-24 by HimGautam