learnsyslab/gym-pybullet-drones #81: How to use multiple process calculate controller?

Wan-ZL, 2022-04-26:
Hi, 

I'm using dozens of drones in simulation, and the "computeControlFromState" takes a lot of time. I tried the Process from multiprocessing but got errors.

So, I would like to know how to make the for loop below run in parallel and speed up the code.
> ctrl = [DSLPIDControl(drone_model=DroneModel.CF2X) for i in range(num_drones)]  # Initialize "num_drones" controllers
> 
> for i in range(num_drones):                                                     # Compute control for each drone
>> action[str(i)], _, _ = ctrl[i].computeControlFromState(control_timestep=env.TIMESTEP, state=obs[str(i)]["state"], target_pos=TARGET_POS)

Thank you.

JacopoPan, 2022-04-27:
Hello @Wan-ZL,

Process or binding a C implementation would be my first guesses to try and speed-up independent computational blocks.
Do you perhaps want to fork the repo, open a PR with your multiprocessing implementation and we can see where the errors are coming from?

Wan-ZL, 2022-05-04:
Hi @JacopoPan,

I forked the repository and here is the link: https://github.com/Wan-ZL/gym-pybullet-drones

Under the examples folder, I created an example called '[fly_parallel.py](https://github.com/Wan-ZL/gym-pybullet-drones/blob/master/examples/fly_parallel.py)' which is the copy of the 'fly.py' but replace the single process PID control part (line 177 in fly_parallel.py) to the multi-process PID control (line 164 in fly_parallel.py).

The code doesn't raise any error but PyBullet GUI is frozen and the pybullet keeps printing the "pybullet build time: Jan 16 2022 01:49:51" as shown below
[image]

So, how to solve this issue? Or do you have other way to make PID control function (action[str(i)], _, _ = ctrl[i].computeControlFromState()) run in parallel?

Thank you!

JacopoPan, 2022-05-15:
Hi @Wan-ZL,

sorry for the delay but I was busy with a few other tasks, the multiple prints of `pybullet build ..` I suspect come from the different processes importing `pybullet` on their own (as it is an import in class `DSLPIDControl but really only for a couple of Euler/quaternion operations at the very beginning and very end: they could could be factored out as well)

My suggestion is to take the following 2 steps:
- debug your (multiprocess, not multi-threaded) parallelism replacing the control method with a dummy function
- re-factor `DSLPIDControl.computeControl()` as purely functional (also returning and re-accepting its internal state)