learnsyslab/gym-pybullet-drones #83: multiagent.py error for more than 2 drone

atharrashno, 2022-05-03:
Hi, Jacopo!, please don't kill me for naive question:|
I am vary new in RL and quad controlling but i want to understanding this by  comparing  3  multi-agent environments :

leaderfolower
flock
meetup

I find out  thesis  are different in computing reward but i cannot see  test_multiagent.py result for NUM_DRONE >=3  because of :

  action = {0: temp[0][0], 1: temp[1][0]}

why do you limit  action for 2 drone(o and 1 th drone)?

JacopoPan, 2022-05-04:
Hi @atharrashno,

If you look at [`multiagent.py`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/experiments/learning/multiagent.py) (where the models/policies used in `test_multiagent.py` are tested) you'll see that the training scheme (central critic and multiple policies) is hardcoded to the 2-agent scenario

https://github.com/utiasDSL/gym-pybullet-drones/blob/7f8d7167bf0046a94259aaaf4fba7163b3ae5563/experiments/learning/multiagent.py#L124

https://github.com/utiasDSL/gym-pybullet-drones/blob/7f8d7167bf0046a94259aaaf4fba7163b3ae5563/experiments/learning/multiagent.py#L272

Note that this is based on the [`centralized_critic_2.py`](https://github.com/ray-project/ray/blob/master/rllib/examples/centralized_critic_2.py) example in `rllib`

atharrashno, 2022-05-17:
> Hi @atharrashno,
>
> If you look at [`multiagent.py`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/experiments/learning/multiagent.py) (where the models/policies used in [...]

tanks a lot jacop

atharrashno, 2022-05-17:
Hi Jacopo!
Thank you for responding, however because I've gone from image processing to this field, I'd want to request that you offer me with a roadmap to help me understand the things you mention in your answer.

atharrashno, 2022-06-10:
Hello jacop!
I altered the hard code from two to three drones after reading the OpenAI paper, but the result was really poor. Now I'd want to implement the concept of piloting four drones in two separate groups, as the performance of the two drones is fairly optimal. Do you believe this concept to be reasonable? Where do you believe this implementation should commence?

JacopoPan, 2022-06-10:
I'd start to think of what the reward function is (and the state-value/q-value functions it would lead to).

atharrashno, 2022-06-14:
I am familiar with the reward function (in 3 Maneuver: leaderfollower,meetup,flock) and the state value function (get Box of 20 values in _getDroneStateVector), but I am unable to locate the q-value function in the gym-pybullet-drone functions.

JacopoPan, 2022-06-14:
Reward, actions, and states exist on the side of the "environment"/MDP (i.e. within `gym-pybullet-drones`, policies (or actors) and value functions (or critics) are models that you want to learn on the agent's side (see the [RL textbook](http://www.incompleteideas.net/book/ebook/node34.html)).

atharrashno, 2022-06-15:
Thank you
I understand the ideas, but to put them into practice, I have to find the code for each model in gym-pybullet-drone and make the changes that are needed.

JacopoPan, 2022-06-15:
`gym-pybullet-drones` classes are environments; policies and value functions reside on the agent side, if you look at the scripts in `examples/learning`, there are agents based on `stable-baselines3` and `ray[rllib]` (but you could implement your own) and you can access/modify the networks used to approximate each function there.

closed 2022-08-03 by JacopoPan

BCWang93, 2023-04-05:
> 

@atharrashno ，how did you change the code from 2 drones to 3 drones?Thanks