learnsyslab/gym-pybullet-drones #84: Confused between simulation freq and physics step arguments in the environments.

HimGautam, 2022-05-10:
Simulation freq and Physics step are the two arguments present in every pybullet-drones environments. But I am not able to understand what they do. 

My requirement is that when I train my reinforcement learning algorithm, the simulation must run as fast as my CPU allows but when I am testing/rendering the environment, the simulation must run in the real time. What should I do? please explain.

JacopoPan, 2022-05-11:
Hi @HimGautam,

I'm sorry this is not more intuitive (we simplified this API in [safe-control-gym](https://github.com/utiasDSL/safe-control-gym), in any subclass of `BaseAviary`:
- argument `freq`, sets `self.SIM_FREQ` and `self.TIMESTEP` (as 1/`freq`), the latter is what goes into Bullet's physics engine as timestep (via `p.setTimeStep()`)
- argument `aggregate_phy_steps`, tells that every call to a `BaseAviary` environment `.step()` method will contain `aggregate_phy_steps` calls to Bullet's physics engine method `p.stepSimulation()`

In short, `freq` sets the physics simulation frequency but `aggregate_phy_steps` tell how much more "coarse" is the `step()` method of a  `BaseAviary` class instance.

Note that you can further refine the number of solver iteration and substeps used by Bullet with method `p.setPhysicsEngineParameter()`

HimGautam, 2022-05-12:
Got it.

closed 2022-05-12 by HimGautam