learnsyslab/gym-pybullet-drones #89: ModuleNotFoundError in ROS2 wrapper node

BeomyeolYu, 2022-05-18:
Hi, thanks for your excellent work.

I'm using Ubuntu 20.04 with ROS2 Foxy Fitzroy on Python 3.8. To use ROS2 python wrapper, I followed the instructions below:
```bash
cd gym-pybullet-drones/ros2/
source ros2_and_pkg_setups.bash
colcon build --packages-select ros2_gym_pybullet_drones
source ros2_and_pkg_setups.bash
ros2 run ros2_gym_pybullet_drones aviary_wrapper
``` 
However, I encountered ```ModuleNotFoundError: No module named 'gym_pybullet_drones'``` error. So, I needed to add the following part to ```aviary_wrapper.py``` and ```random_control.py```:
```bash
import sys
sys.path.append("/home/ben/gym-pybullet-drones")
``` 
But after that I got ```ModuleNotFoundError: No module named 'gym'``` error, thus I had to modify them as follows: [link](https://github.com/BeomyeolYu/gym-pybullet-drones/commit/a0d8aa44892dc6e41adf0820136c11f1347aec59)
```bash
import sys
sys.path.append("/home/ben/gym-pybullet-drones")
sys.path.append("/home/ben/anaconda3/envs/drones/lib/python3.8/site-packages")
``` 
I was wondering if there are other solutions for this.

Best,

JacopoPan, 2022-05-24:
Hi @BeomyeolYu,

just to make sure, did you
```
pip3 install -e .
```
in folder `gym-pybullet-drones/` before using `source`/`calcon`/`ros2`?

BeomyeolYu, 2022-05-25:
Thanks for the reply @JacopoPan,

 Yes, I did ```pip3 install -e .``` first.

- I run the below in a terminal following  ```README.md```:
```bash
$ conda create -n drones python=3.8
$ conda activate drones
$ pip3 install --upgrade pip
$ git clone https://github.com/utiasDSL/gym-pybullet-drones.git
$ cd gym-pybullet-drones/
$ pip3 install -e .

# For ROS2
$ cd gym-pybullet-drones/ros2/
$ source ros2_and_pkg_setups.bash
$ colcon build --packages-select ros2_gym_pybullet_drones
$ source ros2_and_pkg_setups.bash
$ ros2 run ros2_gym_pybullet_drones aviary_wrapper
```
but ```ModuleNotFoundError: No module named 'gym_pybullet_drones'``` error is still occurring. I was wondering if this error only happens to me. If so, it's okay I'll simply use ```import sys```.

- BTW, it seems the above ```README.md``` needs to be slightly modified as:
```bash
$ cd ~/gym-pybullet-drones/ros2/
$ source ~/ros2_foxy/ros2-osx/setup.bash
$ colcon build --packages-select ros2_gym_pybullet_drones
$ source ros2_and_pkg_setups.bash
$ ros2 run ros2_gym_pybullet_drones aviary_wrapper
```
This is because when I run the second line, ```source ros2_and_pkg_setups.bash```, after the first line, ```cd ~/gym-pybullet-drones/ros2/```, I encounter the following error:
```bash
-bash: ./install/setup.bash: No such file or directory
```
The reason seems to be that ```colcon build``` is required to run ```source ./install/setup.bash``` in ```ros2_and_pkg_setups.bash```. What do you think about it?

JacopoPan, 2022-05-25:
Thanks @BeomyeolYu, 

I agree with your comment on the second line of setup.bash and I added a note to the README.md about ignoring the error on first execution (before running `calcon`).
I also updated the `setup.cfg` file to use underscores instead of dashes.
The solution you proposed for the package dependency can work but I will look a bit further into it to avoid such, installation dependent, workaround.

BeomyeolYu, 2022-05-25:
Thanks for the update @JacopoPan,

I think this dependency issue is due to ```conda activate drones``` with ```conda```.

According to [Including a Python module in a ROS2 package](https://answers.ros.org/question/367793/including-a-python-module-in-a-ros2-package/) and [Using Python Packages with ROS 2](https://docs.ros.org/en/foxy/How-To-Guides/Using-Python-Packages.html), they say
- When ROS2 builds the ```src``` files with ```colcon```, it puts the python files into ```workspace/install/lib/python3.8/site-packages/<package_name>```. This folder is considered as the root of python packages in the given ROS 2 package.
- If you use something like ```conda```, it is very likely that the interpreter will not match the system interpreter and will be incompatible with ROS2 binaries.

JacopoPan, 2022-05-25:
Hi @JacopoPan,

Yes, I agree (I couldn't test it today yet because I have to re-install ROS2 on my macOS platform) but I'd prefer not to have 2 separate set of instructions (with and without `conda`) and, if one is to manually add something to the sys.path, I would prefer to make it a one liner with the location of the `conda` packages/modules for the environment (and agnostic to where they are by default in macOS and Ubuntu, instead of the 2 separate imports hardcoded with your home path and the way the folder tree is created for Ubuntu).

JacopoPan, 2022-05-25:
Btw, if you do it before me and create a small patch as a PR, I'd be happy to merge it

BeomyeolYu, 2022-05-30:
Thanks for your time @JacopoPan. That was great.

One thing I'd like to check is whether you have tried your solution on Ubuntu.

I solved ```FileNotFoundError: [Errno 2] No such file or directory``` error and pull requested.

JacopoPan, 2022-05-30:
@BeomyeolYu 

no, I did that on macOS, thanks for the PR! I'm a bit out of the Ubuntu/ROS dev cycle right now so I'm asking @spencerteetaert to look into this over the next few weeks.

closed 2022-07-20 by JacopoPan