learnsyslab/gym-pybullet-drones #91: AttributeError: 'LeaderFollowerAviary' object has no attribute '_agent_ids'

JiahengZeng, 2022-05-26:
When I try to run python3 multiagent.py --act one_d_rpm,  I meet this problem. Can you help me?

ray::PPOTrainer.__init__() (pid=12704, ip=192.168.0.109, repr=PPOTrainer)
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 830, in __init__
    super().__init__(
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trainable.py", line 149, in __init__
    self.setup(copy.deepcopy(self.config))
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 911, in setup
    self.workers = WorkerSet(
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 162, in __init__
    self._local_worker = self._make_worker(
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 567, in _make_worker
    worker = cls(
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 507, in __init__
    check_env(self.env)
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/utils/pre_checks/env.py", line 63, in check_env
    check_multiagent_environments(env)
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/utils/pre_checks/env.py", line 196, in check_multiagent_environments
    sampled_obs = env.observation_space_sample()
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/env/multi_agent_env.py", line 210, in observation_space_sample
    self._check_if_space_maps_agent_id_to_sub_space()
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/env/multi_agent_env.py", line 345, in _check_if_space_maps_agent_id_to_sub_space
    and set(self.observation_space.spaces.keys()) == self.get_agent_ids()
  File "/home/zjh/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/env/multi_agent_env.py", line 229, in get_agent_ids
    if not isinstance(self._agent_ids, set):
AttributeError: 'LeaderFollowerAviary' object has no attribute '_agent_ids'

JiahengZeng, 2022-05-27:
I solved this after downgrading ray[rllib] to version 1.9

JiahengZeng, 2022-05-27:
In addition, I would like to ask how to rotate the viewing angle when rendering？
And why is half of the text in my GUI missing？
[image]

JacopoPan, 2022-05-28:
@JiahengZeng 

The view that the GUI is initialized with is set in these lines:
https://github.com/utiasDSL/gym-pybullet-drones/blob/4f81a89adda2a16604c441d811367f0ca8092c5d/gym_pybullet_drones/envs/BaseAviary.py#L178-L191

To move the view during the preview, look at this answer on Bullet's forums:
https://pybullet.org/Bullet/phpBB3/viewtopic.php?p=41476&sid=4f0297813ba516fe2f1b18fc5f10f45c#p41476

JiahengZeng, 2022-05-29:
@JacopoPan 
Thank you very much, you are so kind.