learnsyslab/gym-pybullet-drones #92: Obs State Vector Reference Frame Confusion

imamim, 2022-05-27:
Hello, thank you for help and all advice. 

I work with your repositories, but I got a question.

Are the angular velocity values ​​returned by the `_computeObs()` function relative to the body frame or World Frame ? I try to test this given angular velocity. But I can't get any result. I need help for this.

Also thank you for answer.

JacopoPan, 2022-06-06:
`_computeObs()` normally leverages `_getDroneStateVector()` that in turn uses:

https://github.com/utiasDSL/gym-pybullet-drones/blob/d64001bb4b724f7bc060924dbe4bda8c6c2aaf51/gym_pybullet_drones/envs/BaseAviary.py#L504-L514

For detailed information on each of the PyBullet APIs, you can refer to the [Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.4vxw9j7piyjd), for `getBaseVelocity` specifically:
> This returns a list of two vector3 values (3 floats in a list) representing the linear velocity [x,y,z] and angular velocity [wx,wy,wz] in Cartesian worldspace coordinates.

imamim, 2022-06-06:
Firstly thank you for the response. That's what I expected too, but I couldn't find enough information in the Quickstart Guide. But now it's completely understood. Thank you :) Good luck with

closed 2022-06-06 by imamim