learnsyslab/gym-pybullet-drones #98: Single agent not learning to hover using velocities as actions

ranzuh, 2022-06-17:
Hi,

I'm trying to conduct experiment where I train a neural network policy with PPO to control the drone with velocity commands. I'm using experiments/learning/singleagent.py with following command ```python singleagent.py --env hover --algo ppo --obs kin --act vel --cpu 1```

PPO is not able to learn the task even if I increase the time-steps. With the default RPM-actions it learns fine. My intuition says that the task should be even easier with the velocity actions, so it's confusing why PPO can't learn it. Is my intuition wrong and the task is actually more difficult with velocity actions or might there be something weird going on in the implementation of velocity actions?

hidal00p, 2022-06-17:
Hello!

I am not the developer of this project, but I would like to share my thoughts on your issue, hope you would not mind.

> My intuition says that the task should be even easier with the velocity actions

Could you please elaborate, why exactly you think this way?
I would say that a contrary opinion makes more sense.

This is what ActionType enum says in the BaseSingleAgentAviary.py:
[image]

So the default action specified in singleagent.py is actually ONE_D_RPM, which is a 1D uniform input signal to all motors
[image]

While the velocity(VEL) action is a vector of 4 distinct numbers sent to each motor individually.

Due to the fact that quad can be considered as a symmetrical structure applying a uniform rpm signal would generate only thrust  in the direction opposite to gravity, thus the only thing your policy needs to figure out is how much thrust exactly it needs to generate to reach a desired altitude and keep it without unnecessary acceleration upwards or downwards, or not to remain stationary on the ground.

Whereas in case of individual signal supply to every motor, your policy needs to understand how to properly sync the motors, which also introduces problems with orientation on top of the aforementioned issues. It makes problem more multidimensional, thus increasing level of complexity.

This would be my reasoning for why it is harder for PPO to capture a solution. Probably it needs more time, or other more sophisticated tuning may be required , who knows.

Let me know what you think. Cheers.

ranzuh, 2022-06-18:
Hello, thank you for the reply! Yeah, I didn't notice that it was using the ONE_D_RPM and that it is only 1 component. It makes sense that 4D is harder than 1D. I actually got it to learn better when I supplied PPO with `ent_coef=0.01` (default=0) parameter to make it explore more. I will close this issue as this seems not to be a problem in environment implementation but in tuning the learning algorithm.

closed 2022-06-18 by ranzuh

JacopoPan, 2022-06-19:
Great conversation @ranzuh @hidal00p :D

@spencerteetaert to consider for the FAQ/discussion page

dee6600, 2022-11-18:
thank you for this talk