learnsyslab/gym-pybullet-drones #104: How to create new drone model?

zcase, 2022-07-22:
I would like to create a new drone model. Do you have a tutorial or something on how to create the new model based off the data sheet like you did for the based drone model?

JacopoPan, 2022-07-26:
Hello @zcase 

it should be as simple as creating a new URDF file like this one https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/assets/hb.urdf and adding it to the enum class https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/utils/enums.py

If you do it in a personal fork, you can add a PR to this repo and I can help you there.

Ziad-Shoeib, 2026-02-14:
Hello @JacopoPan,
Thanks a lot for your help, I appreciate it.

Yes, I am controlling the rpm actions, and built the drone the same way as the `cf2x.urdf`,  just scaling up dimensions and properties:
```xml
  <properties arm="0.125000" kf="1.813450405e-08" km="1.686996460e-10" thrust2weight="4.693468" max_speed_kmh="36" gnd_eff_coeff="11.36859" prop_radius="0.063500" drag_coeff_xy="9.11e-6" drag_coeff_z="1.023e-5" dw_coeff_1="2267.18" dw_coeff_2=".16" dw_coeff_3="-.11" />
```

From what I understand, if instead I wanted to work with pid actions, I need to tune `DSLPIDControl` to work well with my URDF (I actually tried the same values of cf2x, and the performance was not good in `pid.py`, for example).

Another side question, I had in mind, in building any other drone urdf in `gym-pybullet-drones`, the same style as `cf2x.urdf`. For the aerodynamic parameters `drag_coeff_xy="9.11e-6" drag_coeff_z="1.023e-5" dw_coeff_1="2267.18" dw_coeff_2=".16" dw_coeff_3="-.11"`, if the drone I am working with does not mention them at all, is it valid just scale them up by the same factor of the drone size to cf2x size. (My work will be concerned with aerodynamic disturbances that is why I care about these parameters specially). Also, I looked at similar drones in size in PX4 models with gazebo, but almost all of what I found just mentions `rotor_drag,` not the body drag.