learnsyslab/gym-pybullet-drones #105: Hummingbird velocity controller

pladosz, 2022-08-01:
Hi,

I am trying to train using hummingbird model while controlling velocity. However, It seems right now hummingbird only supports position controller. Is that correct or am I missing something?

JacopoPan, 2022-08-02:
Hi @pladosz 

Yes, you are right in that `VelocityAviary` theoretically supports the Hummingbird drone model but the controller it uses by default, `SimplePIDControl`, does not support velocity inputs. In principle, you could circumvent the issue by adapting `DSLPIDControl` (which is parametrized for the Crazyflie), extending `SimplePIDControl`, or modifying `_preprocessAction()` in `VelocityAviary` to set a `target_pos` derived from the desired velocity.

I will modify `VelocityAviary` to raise an exception when `drone_model == DroneModel.HB` but if this is something you are interested in contributing to, please feel free to create a fork and PR. 🙏

pladosz, 2022-08-03:
Thank you for the extensive reply. I think raising exception is a good idea. I will try to play with DSLPIDControl when I have time and make PR if successful. 

In the meantime are you aware of any bigger model (apart from hummingbird) which is compatible?

JacopoPan, 2022-08-03:
I think it could suffice to adjust the values in `__init__()` DSLPIDControl.py (mostly the PWM to RPM factors and the noting that `HB` is in the same + configuration as `CF2XP` to chose the `MIXER_MATRIX`).

pladosz, 2022-08-04:
Thank you for the advice. I found it easier to just retune hummingbird's PID controllers to work for velocity while copying some elements from DSL controller. Once I have time to wrap up the code and prepare PR I will do so.