learnsyslab/gym-pybullet-drones #109: How to collect data?

HimGautam, 2022-08-15:
Hi @JacopoPan ,
I wanted to do a project using deep learning. For this project I wanted offline data containing states, actions, next_states. So is there an environment in which I can directly control the quadcopter from my keyboard?

JacopoPan, 2022-08-15:
Hi @HimGautam, 
the functionality you are looking for is not currently implemented. It is technically possible to do and not overly complicated (e.g., starting from a class like `VelocityAviary` that already contains a PID controller and using something like `pygame`). I would have one major doubt: do you really need human interaction (or could you just script the behaviour beforehand)? Human-in-the-loop data collection might be to great a bottleneck for many deep learning tasks.

HimGautam, 2022-08-15:
Human in the loop is not required. It is just that a diverse dataset is needed.

JacopoPan, 2022-08-15:
You could just use [`fly.py`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/examples/fly.py) with a single drone (`DEFAULT_NUM_DRONES = 1`) and change `TARGET_POS = np.zeros((NUM_WP,3))` to the trajectory you want.

HarshilNaikk, 2022-10-10:
@JacopoPan Can you elaborate a bit more about using your own trajectory? Cause when I try to use my own trajectory using algorithms like RRT*, the drone just crashes every time. I see variation in its behaviour when I change the parameters "NUM_WP, simulation_freq_hq, control_freq, and aggregate_phy_steps". Can you expand a bit more about how you set and tune these parameters? I have a feeling NUM_WP is not just equal to the number of waypoints in your trajectory. It might have more to do with the actual way the waypoints are passed to the controller as well?

JacopoPan, 2022-10-10:
@HarshilNaikk 

editing `fly.py` still relies on `[DSLPIDControl.py](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/control/DSLPIDControl.py)` whose parameters are tuned to track a trajectory of (at least) positions (and possibly velocities) that are not too far from one another.

If you are only generating sparse waypoints with a planner, you'd need to interpolate them, e.g. with a polynomial fit.
Otherwise, if you have waypoints far apart, the PID controller will try to "jump" between them and lose control.