learnsyslab/gym-pybullet-drones #112: How hard is to migrate an ideally trained model into a real Crazyflie drone?

camillo2008, 2022-09-20:
Maybe my question is silly, I am sorry if this is the case.

I am wondering how hard is to migrate a trained model with SB3 or RLlib to a real Crazyflie?
And how accurate this will be?
Do I need to use ROS2, or I can use python as well

JacopoPan, 2022-09-20:
Hello @camillo2008 

migrating a model to a real-life Crazyflie require to bring the weights of a (small) trained network onto the Crazyflie firmware. It has been done before for various applications (https://arxiv.org/pdf/2103.01885, https://arxiv.org/pdf/2201.01369.pdf) but it's non-trivial because of the on-board computational constraints.

If you are looking at a project fo off-board control that integrates 1-to-1 with the real quad rotors and [Crazyswarm](https://crazyswarm.readthedocs.io/en/latest/), you should look at our [IROS 2022 competition codebase](https://github.com/utiasDSL/safe-control-gym/tree/beta-iros-competition) ([video](https://www.youtube.com/watch?v=PwphA_jsNKw))

camillo2008, 2022-09-25:
Thanks so much for the guide, the competition seems to be a good place to start!

closed 2022-09-25 by camillo2008