learnsyslab/gym-pybullet-drones #114: replay script: test_multiagent.py gives error on multiple machines (TypeError: can't convert np.ndarray of type numpy.object_.)

jackvice, 2022-10-07:
checkpoint.txt seems good and is found but still this error.  Tried on two different machines, different GPU's.
Traceback (most recent call last):
  File "test_multiagent.py", line 256, in <module>
    agent.restore(checkpoint)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trainable.py", line 462, in restore
    self.load_checkpoint(checkpoint_path)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 1629, in load_checkpoint
    self.__setstate__(extra_data)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 2110, in __setstate__
    self.workers.local_worker().restore(state["worker"])
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 1342, in restore
    self.policy_map[pid].set_state(state)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/policy/torch_policy.py", line 715, in set_state
    optim_state_dict = convert_to_torch_tensor(
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/utils/torch_utils.py", line 157, in convert_to_torch_tensor
    return tree.map_structure(mapping, x)
  File "/home/jack/.local/lib/python3.8/site-packages/tree/__init__.py", line 430, in map_structure
    [func(*args) for args in zip(*map(flatten, structures))])
  File "/home/jack/.local/lib/python3.8/site-packages/tree/__init__.py", line 430, in <listcomp>
    [func(*args) for args in zip(*map(flatten, structures))])
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/utils/torch_utils.py", line 151, in mapping
    tensor = torch.from_numpy(np.asarray(item))
TypeError: can't convert np.ndarray of type numpy.object_. The only supported types are: float64, float32, float16, complex64, complex128, int64, int32, int16, int8, uint8, and bool.
Destroy EGL OpenGL window.

JacopoPan, 2022-10-10:
Hi @jackvice 

thanks for the potential bug reporting can you tell me

camillo2008, 2022-10-19:
I am facing the same issue, do you have found any solution?

1. Commands in sequence: 1) python3 multiagent.py wait until completed 2) python3 test_multiagent.py --exp [checkpoint folder], this command cause the error.

2.  Ubuntu 20.04.4 LTS

4. packages version:
    - torch: 1.12.1
    - ray: 1.9.0
    
5. All terminal output
`[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
2022-10-19 20:31:14,963	INFO ppo.py:166 -- In multi-agent mode, policies will be optimized sequentially by the multi-GPU optimizer. Consider setting simple_optimizer=True if this doesn't work for you.
2022-10-19 20:31:14,964	INFO trainer.py:743 -- Current log_level is WARN. For more information, set 'log_level': 'INFO' / 'DEBUG' or use the -v and -vv flags.
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
Traceback (most recent call last):
  File "/home/cam/Desktop/Tutor/SVS/gym-pybullet-drones/experiments/learning/test_multiagent.py", line 256, in <module>
    agent.restore(checkpoint)
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trainable.py", line 462, in restore
    self.load_checkpoint(checkpoint_path)
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 1629, in load_checkpoint
    self.__setstate__(extra_data)
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 2110, in __setstate__
    self.workers.local_worker().restore(state["worker"])
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 1342, in restore
    self.policy_map[pid].set_state(state)
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/policy/torch_policy.py", line 715, in set_state
    optim_state_dict = convert_to_torch_tensor(
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/utils/torch_utils.py", line 157, in convert_to_torch_tensor
    return tree.map_structure(mapping, x)
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/tree/__init__.py", line 430, in map_structure
    [func(*args) for args in zip(*map(flatten, structures))])
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/tree/__init__.py", line 430, in <listcomp>
    [func(*args) for args in zip(*map(flatten, structures))])
  File "/home/cam/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/utils/torch_utils.py", line 151, in mapping
    tensor = torch.from_numpy(np.asarray(item))
TypeError: can't convert np.ndarray of type numpy.object_. The only supported types are: float64, float32, float16, complex64, complex128, int64, int32, int16, int8, uint8, and bool.

Process finished with exit code 1
`
The problem is that np.asarray() encounter None

JacopoPan, 2022-10-19:
The line in this repo throwing the error seems to be
```
gym-pybullet-drones/experiments/learning/test_multiagent.py", line 256, in 
agent.restore(checkpoint)
```

i.e.
https://github.com/utiasDSL/gym-pybullet-drones/blob/2b09b34d2309e6aea4db91dfd2cc9492ca8fe67a/experiments/learning/test_multiagent.py#L252-L256

Have you checked the agent has been trained/saved/the correct path to re-load it is being passed?

camillo2008, 2022-10-19:
yeah, that's the line, and yes i have checked the path,  for example now i am passing the real path that generate the error for me, this is the string inside checkpoint visualized with the debugger:'/home/cam/Desktop/Tutor/SVS/gym-pybullet-drones/experiments/learning/results/save-leaderfollower-2-cc-kin-one_d_rpm-10.19.2022_21.15.40/PPO/PPO_this-aviary-v0_6d227_00000_0_2022-10-19_21-15-43/checkpoint_000001/checkpoint-1'
Maybe you see something odd, i don't know really well ray, but to me it seems correct, the file exist, call command cat with this same path does return something.

camillo2008, 2022-10-19:
Maybe the problem is not with the loading function but with the saving one, i have tested the same thing with the single agent and work, but is based on sb3

JacopoPan, 2022-10-19:
@jackvice @camillo2008 

This seems to work on my end, this are some of the main packages in my conda environment
```
(drones) jacopo@Jacopos-MacBook-Pro learning % conda list
# packages in environment at /Users/jacopo/opt/anaconda3/envs/drones:
#
# Name                    Version                   Build  Channel
...
gym                       0.21.0                   pypi_0    pypi
gym-pybullet-drones       0.0.3                    pypi_0    pypi
...
matplotlib                3.5.1                    pypi_0    pypi
...
numpy                     1.22.3                   pypi_0    pypi
...
pandas                    1.4.2                    pypi_0    pypi
pip                       22.1.2                   pypi_0    pypi
pybullet                  3.2.5                    pypi_0    pypi
...
ray                       1.9.0                    pypi_0    pypi
...
scikit-image              0.19.2                   pypi_0    pypi
scipy                     1.8.0                    pypi_0    pypi
stable-baselines3         1.5.0                    pypi_0    pypi
...
tensorboard               2.9.1                    pypi_0    pypi
tensorboard-data-server   0.6.1                    pypi_0    pypi
tensorboard-plugin-wit    1.8.1                    pypi_0    pypi
tensorboardx              2.5                      pypi_0    pypi
torch                     1.11.0                   pypi_0    pypi 
...
```
maybe try `pip install torch==1.11.0 torchvision==0.12.0 torchaudio==0.11.0
` or `conda install pytorch==1.11.0 torchvision==0.12.0 torchaudio==0.11.0 -c pytorch` (see [this page](https://pytorch.org/get-started/previous-versions/))

JacopoPan, 2022-10-19:
@camillo2008 let me know if that solves it and I will update [`pyproject.toml`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/pyproject.toml)

camillo2008, 2022-10-21:
Yes, it seems that torch is the problem, using: `pip install torch==1.11.0 torchvision==0.12.0 torchaudio==0.11.0` as you suggested resolving the issue, though I had to recreate the sample, using the one created with torch 1.12.1 still crash even when loaded with torch 1.11.

Thanks, for me this issue is resolved

closed 2022-10-21 by JacopoPan

jackvice, 2022-10-23:
Thank you!  Reverting to torch 1.11 does work but only for 2 drones.  When I try 3 or 4 drones, it get:
File "test_multiagent.py", line 322, in <module>
    state= np.hstack([obs[j][0:3], np.zeros(4), obs[j][3:15], np.resize(action[j], (4))]),
KeyError: 2

I tested on two different GPU machines, one with torch 1.11 and the other torch 1.10.

JacopoPan, 2022-10-23:
Hi @jackvice,

that is expected, if you read lines 63-137 in [`multiagent.py`](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/experiments/learning/multiagent.py) you'll see the centralized critic is only designed to account for 2 agents.