learnsyslab/gym-pybullet-drones #119: How to control drone using my desired roll pitch yaw?

Jay1470, 2022-10-25:
Hi @JacopoPan ,
I am currently trying to implement visual servoing using your simulator. My application requires that I control the quadrotor using the roll pitch yaw obtained from my equations. Can you please tell me how can I control the quadrotor using my roll, pitch yaw commands.

JacopoPan, 2022-10-27:
Roll pitch yaw or roll pitch yaw *rates*? You'll need to write your own controller for that OR try to use a Python wrapper (like [`pycffirmware`](https://github.com/utiasDSL/pycffirmware) of the Crazyflie firmware (that has a roll pitch yaw thrust interface in `cmdVel`)

DS3a, 2023-04-21:
Hi @JacopoPan ,
I am unable to find the control scheme you mentioned in `pycffirmware` (roll, pitch, yawrate, thrust).
[attitude_pid](https://github.com/utiasDSL/pycffirmware/blob/main/src/modules/src/attitude_pid_controller.c) has roll, pitch, yaw, rollrate, pitchrate, yawrate. I don't want to control the rollrate, and pitchrate, and the yaw. And, I don't think setting them to zero will work. Moreover there is no mention of thrust.

I also found [controller_pid](https://github.com/utiasDSL/pycffirmware/blob/main/src/modules/src/controller_pid.c), this has roll, pitch, yaw and thrust. everything works, except for yawrate. Is there anything I am missing?